Ralim/IronOS #450: TS100_TR.hex v2.06-RC4 does not work

koklica, 2019-02-08:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-02-09:
@koklica 
You have provided no information in this report.
I will close this after 24 hours if information is not provided.

koklica, 2019-02-09:
Turkish language file missing is not working

Ralim, 2019-08-27:
Please try RC5, and re-open if still having issues. Would love a photo of which screen has the issues.

closed 2019-08-27 by Ralim