Ralim/IronOS #451: SainSmart Pro32 (Ts100) - 100 Degree C Tip offset?

Bulltac, 2019-02-10:
Please help with this issue it has crippled my brand new iron

* **I'm submitting a ...**
  - [* ] Bug report
 

* **Do you want to request a *feature* or report a *bug*?**
Report a Bug

* **What is the current behavior?**
Tip reads within 10 degree C of true temp when powered via USB, however when it is plugged in VIA 19V 2A DC supply it reads 130C at room temperature. Problem is the same with multiple tips. 

  - Firmware Version: 2.03
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 19V 2A

When powered on via USB it reads close to correct, when powered by the power supply it reads 130C at idle. Please help I am at my wits end and I cant even seem to put the original firmware back on

Bulltac, 2019-02-10:
Also tried Flashing Version 2.05.1 and have the exact same problem... if you dont have any ideas about this would you perhaps know where to get the Minware firmware that came on the Iron?

Ralim, 2019-02-10:
Can you provide all of the readouts from the debugging menu under both conditions.

There is very little difference between both power measurement modes unless your MOSFET / driver circuit is leaky.

Bulltac, 2019-02-10:
Do you want the readouts from both Version 2.05.1 and the Minware to compare?

Ralim, 2019-02-10:
Miniware does not provide these readouts.

Bulltac, 2019-02-10:
Ok I will get them as soon as i get back home this evening. Thank you

Bulltac, 2019-02-10:
Im sorry i feel very stupid for not being able to figure this out but how do I access the debug read-outs? I have reflashed from minware back to 2.05.1 (same problem occuring) but i cannot seem to access the debug read out... I apologize for taking up this much of your time

Bulltac, 2019-02-10:
Think I got it figured, Are these the values you need?

HEAP: 01416
HWMG: 00650
HWMP: 00482
HWMM: 00424
Move: 00095
Rtip: 01192
Ctip: 00137
Vin: 19.0

Ralim, 2019-02-11:
Yep those are the values. Can you confirm if it was around 14C ambient when you wrote those out ?

Bulltac, 2019-02-11:
It was closer to 18C ambient but close enough. I will get the values on 5V for you

Bulltac, 2019-02-11:
Values when plugged into USB (5V)

HEAP: 01416
HWMG: 00650
HWMP: 00482
HWMM: 00424
Move: 00095
Rtip: 01051
Ctip: 00052
Vin: 00.0V

Ralim, 2019-02-11:
Ctip is the tip temperature in X10 format, so it should read ~ambient times ten. For both of these its off, but not drastically.
What readings do you get for Than (Handle temperature) ?
Rtip is the raw adc reading of the temp, 1000 at idle is about right for 15C so curious to know whats messed up. 

Can you tried a factory reset in the menu ?

Bulltac, 2019-02-11:
I have tried a factory reset several times on 2 different firmware versions and plugged into the USB and again plugged into the DC power supply. I have checked the debug menu several times and there is no "Than" parameter showing up as far as i can tell.... perhaps i am looking in the wrong place but i have reviewed the debug menu on the iron several times

Ralim, 2019-02-12:
I'll have to make a build with than in it for ts100, sorry.

Just to confirm, when you are using the 19V DC supply, the tip reads consistently high compared to 5V? Have you tried other DC sources ?

The readings you have so far from the debugging menu look completely fine.

Which tip do you have selected in the advanced menu ?

Bulltac, 2019-02-12:
Yes, So when the iron is plugged in via USB, (multiple cords, ports and computers) it reads between 40 and 50C tip temp. When it is plugged in via multiple 19V/2A supplies AND when it is plugged into 4s or 5s battery it consistently reads 130C-140C at idle. 

As far as i know the tip i have selected is the B2 which is the tip I have had in 90% of the time, I did try switching tips to the BC-2 tip and switched the tip in the advanced menu but got exactly the same problem.. 
In another post a couple of people had similar issues that were fixed by opening up the iron and touching up the solder on the temp sensor but i have inspected the senser multiple times under microscope and as far as i can tell it is soldered to the board correctly.

I want to thank you again for all your help man, I wish I could have bought this Iron from you in some way because just about everyone I know buys this iron specifically to run your firmware..

Bulltac, 2019-03-01:
UPDATE: SOLVED

Just want to update you on this situation as I dont want you wasting your time on this. I ended up having the seller replace the item and it showed up yesterday. (still sainsmart version) I tested it to see if it worked and then flashed your firmware right away. The Iron and the firmware are both working exactly as intended. Tip temp is accurate and consistent and I have built most of a quadcopter without issue so I am calling this one as hardware failure.... I read a few people with a similar issue reflowed their boards and got it to work (both with miniware and sainsmart versions) so again I believe this is proof of a hardware problem.

I want to thank you for the support and I was wondering where I might make a donation to your efforts?  I feel like Ive gotten better support here than I have from companies I spend thousands on.
I will leave closing the concern to you as I am not sure if you have looked into this issue and found something that I missed.

Cheers,
Logan

dhiltonp, 2019-03-14:
Here's ralim's paypal :) https://www.paypal.me/RalimTek

closed 2019-12-29 by Ralim