Ralim/IronOS #453: Instructions to flash via SWD

FFY00, 2019-03-01:
@Ralim can you add instructions to flash via SWD? I was able to connect my ST-Link to the board with openocd but I don't know the memory addresses to write to. Can you please share the commands to dump and write the firmware? Thanks

Ralim, 2019-03-03:
What is the case you are trying to recover?
You can write the backup hex file which will write the stock bootloader + you can then flash this firmware over the top?
The hex files contain the addresses in themselves so they should be just flashable as they are.

You will need some form of bootloader on the device as the firmware is compiled expecting the vector table to be offset.

FFY00, 2019-03-03:
I currently get no display with your builds but it works on stock fw. Maybe something is wrong with the bootloader? I wanted to flash it directly to see if it helps.

Where is the backup hex?

FFY00, 2019-03-12:
So, I was able to flash your backup but I get the same behavior. I also tried making a version with a slowed down I2C bus but that didn't seem to have any effect. Do you have any other ideas?

To make the slowed down version I changed https://github.com/Ralim/ts100/blob/master/workspace/TS100/src/Setup.c#L234 to `60000`. Do you think that wasn't enough?

Ralim, 2019-03-14:
Can you confirm if you have a ts100 or ts80, and what board revision it is ?

Ralim, 2019-03-14:
I haven't had this issue be reproducible on any of my units, but if you can get the st-link connected while it's assembled enough to test  it's possible to run the code from Atollic so you can debug and check what's failing.

FFY00, 2019-03-14:
I have a TS100. That might be a bit tricky since I have to solder cables there but it might be duable.

Ralim, 2019-12-29:
You can probably just connect the SWD directly to the module, and then dump the flash back out and check against the hex file you flashed to validate the bootloader is writing it correctly.

You could also then attach a debugger with it outside of the iron and check its not jumping to a hardfault.

closed 2020-05-29 by Ralim