Ralim/IronOS #455: Custom firmwares do not work with Baseus power bank, stock firmware does

salavatkhafizov, 2019-03-13:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report  bug

* **What is the current behavior?**
TS80 with D25 tip and *custom* firmwares (both RC and stable) does not work out of the Baseus  power bank (model Baseus Thinnest Digital Dual Output Power Bank,  which equipped with QC 3.0 output port capable of 9V 2A delivery. Power bank shows only 5V output on QC 3.0 port. Using stock firmware (1.07) it works just fine, using the same power bank, 9V output is shown on power bank and TS80.

* **What is the expected behavior?**
TS80 should work normally with 9V output from the above mentioned power bank.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Install any custom firmware.
2. Connect to the above power bank
3. Install stock firmware
4. Connect to the above power bank.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
Use power bank, which is convenient and meant for the use with TS80

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x (see above)
  - PCB Version: (1/2) (not sure)
  - Power Supply (Voltage and Current Rating) :
 Charger 9V 2A, 12V 1.5A (provided with TS80).
Power bank has the same output parameters.

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-03-13:
This most likely means that the power bank does not follow the specifications that i follow for QC.
Any chance at all you are possibly able to get an oscilloscope or analog logic analyzer capture of the data pins on the usb port during negotiation ?

whitehoose, 2019-03-13:
Te QC3 "standard" is anything but  It's too late now, but for future purchases a little time spent with google should confirm powerbanks that will work consistently.
Xiaomi MI 2 – 10000 mAh Power Bank is one that I can confirm

If you wade through the main support thread https://github.com/Ralim/ts100/issues/349
you'll see that you're not alone.
possibly one of the better solutions is to use a AOKoda Lipo to USB Power Converter QC3.0 Adapter as a QC3 power conditioner, unfortunately it requires a 7-27v source to power it.

This is how I put together an adapter that works with any source from 3-27v 
https://github.com/Ralim/ts100/issues/349#issuecomment-463796747

Its cheap enough to build and while it's a bit clumsy - it means you can be confident you'll get the iron working whatever you are faced with

salavatkhafizov, 2019-03-14:
@whitehoose  It is understandable that not every charger/bank follows QC standard, but mine works with stock firmware, which means (I hope) it can work with custom firmware too.

@Ralim You mean negotiation during the initial connect? I do not see any initialisation on the power bank screen (like changing from 5V to 9V).. I have a scope, very simple one, up to 10 Mhz  bandwidth, but it is very basic one.

Ralim, 2019-03-14:
A basic scope is fine, QC is negotiated on the order of milliseconds, even a 10khz scope would be enough :)

I've tried to keep my firmware to follow the QC standards that I can follow, but have had a very mixed results with some power banks that implement weird timings.

If you can possible get rough voltage captures if the two USB data pins duringhe first like 2 seconds it would be a good help.

salavatkhafizov, 2019-03-14:
I will try to implement some sort of USB wiring/connector where I can hook a scope during negotiation process with stock one and custom firmware to post these results (have to find parts or disintegrate existing cable )))).

salavatkhafizov, 2019-04-15:
@Ralim 
I've got logic analyser and run some recording with stock and custom FW. Please let me know what format you need of LA recording.

salavatkhafizov, 2019-04-15:
[customFW.txt](https://github.com/Ralim/ts100/files/3081325/customFW.txt)
[stockFW.txt](https://github.com/Ralim/ts100/files/3081326/stockFW.txt)

salavatkhafizov, 2019-04-15:
Screenshot capture
[image]

INGY23, 2019-05-16:
I am also having this issue with a Anker PowerCore Speed 20000. It works fine on stock both 1.06 and 1.07. On 2.06-RC4 it will only run at 5V. With the wiki listing "Anker PowerCore II Slim 10000 Powerbank" I was rather hoping it would work with the 20000 as well.

salavatkhafizov, 2019-05-21:
@INGY23 Ralim mentioned that he tries to cope with different QC protocols, but (my understanding) they are not uniform, so that I have 2 firmwares working and not working on 2 separate power adapters and 1 power bank.

whitehoose, 2019-05-21:
@INGY23 Different in this case has many faces.  Qualcomm wrote the book on QC - which it seems took off bigtime, unfortunately they wanted paying for using it.  Pointing no fingers the majority of the people set to benefit are all in one area and shares the same philosophy as scrooge MacDuck, They won't pay for anything.  So they reverse engineered some gadgets.  Some of the people were very clever and some weren't ... and so not everyone is starting from the same place or using the same rules.

Look up this thread to my post dated 13Mar

> This is how I put together an adapter that works with any source from 3-27v
> #349 (comment)
[https://github.com/Ralim/ts100/issues/349#issuecomment-463796747](url)

Aokoda have produced a gadget to use (mainly) with LiPo batteries to give (as far as I can tell) a stable and compatible QC3 output. cost £14 ish
I combined a buck boost converter (with a voltage dislay) with a XT60 connector to get me a 14v source to connect into (I mainly use 4s batteries - but any USB powerbank works well too).
I can confirm it will work with ANY power source from 1x aaa through 5v USB2 to a 27v bank of lipo batteries.
Now I can connect anything from 1x aaa upwards and power any QC3 device I've tried.

OK it's a niggle - but it's in the spirit of hacking and customising. 
Once you've made the leads up all you need are the connectors that allow you to plug in your power sources.  I already had a selection of jacks and battery plugs.  Just needed an adater for the Aokoda

Got a nice potting box 50mm x 80mm x30mm and that does the job.

Ralim, 2019-12-29:
@salavatkhafizov 

Following up on this, do you by chance have time to test this on the latest release, as it implements the QC2 style of 12V rather than QC3 style

salavatkhafizov, 2020-01-13:
@Ralim 
Sorry for the delay in replying.

I checked with Baseus power bank, unfortunately, it does not, even at 5V - heating very slowly, after trying to solder temperature goes even down and does not heat after all.

12V performance from PS outlet is excellent, way better than stock one, thank you very much.

drewzh, 2020-03-23:
> I am also having this issue with a Anker PowerCore Speed 20000. It works fine on stock both 1.06 and 1.07. On 2.06-RC4 it will only run at 5V. With the wiki listing "Anker PowerCore II Slim 10000 Powerbank" I was rather hoping it would work with the 20000 as well.

I just purchased this power bank also and found that it doesn't work. Did you ever manage to find a workaround or did you purchase another battery? If so, which battery did you choose?

whitehoose, 2020-03-23:
@salavatkhafizov drewzh
Look at Look up my threads dated 13Mar & 21 May
You could do worse than consider 4s lipo batteries, they are cheap and easy to work with, along with an AOKoda adaptor I've had no problem with compatibility or falling asleep.  Powerbanks are built for 'phones, read the blurb - I've never seen one say anything about soldering irons ... or any other usb device that only needs a nudge most of the time - They aren't designed to provide a trickle of low current (without falling asleep) it's all about throwing a huge chunk of pixies at a device that needs charging, then switch off the power so you can save what's left for later.

salavatkhafizov, 2020-03-23:
@whitehoose You are probably right, but despite of all your logic, stock firmware does work on power banks, not only my case.

whitehoose, 2020-03-23:
You are not right :grinning: neither firmware works with every qc3 power source - look round the forums - everyone has problems, QC3 has been reverse engineered independently by a selection of manufacturers with the result that there are no guarantees which powersource  (this affects powerbanks, wall warts and adaptors etc across the board). works with devices (phones have a limited repertoire - all they need is to charge a battery - no complex switching).
My (and others) idea of a workround in this case separates the QC3 from the power (I originally plumped for ISDT BattGo BG-8S Smart Battery Checker Balancer Receiver Signal Tester Quick Charge Function from banggood (but they are 4x as expensive compared to the AOKoda).
Once you find a formula its a case of sticking to it and a known adapter then allows you buy and use any battery brand available without the lottery.  I carry 5 or 6 in my tools .... all work
As I said Power banks aren't an ideal solution for the type of use we need.
I have cheap USB2s as well as a qc3 wall wart that refuse to work with low currents (my qc3 PB has an override) without introducing a power LED just to keep the pixies dancing - sort of defeats the point of using low current kit..

Ralim, 2020-05-29:
@salavatkhafizov This may be improved in the latest release so would be curious if you could re-test when you have time 😅

salavatkhafizov, 2020-05-29:
@Ralim Tested after it was released.. Unfortunately it works only with 5V, not with 9V.

Ralim, 2020-05-29:
Ah bummer :( 
Dont really want to buy one to test against, but i might if i see one go on sale.

Do you know exactly which model this is?

salavatkhafizov, 2020-05-29:
If you can explain how to find model)) or you are talking about Baseus power bank? BS-P10KQ02

Ralim, 2020-05-29:
Ah yes I did mean the power bank.
If it comes up on sale ill pick one up to test against. (One can never have too many power banks).

Ralim, 2020-11-01:
@salavatkhafizov Is this still broken for you? There have been _some_ changes to QC in recent releases that _may_ help.

I have been doing another round of QC testing on more devices recently and have not run into any that have failed recently for me :/

salavatkhafizov, 2020-11-01:
@Ralim unfortunately no :( installed latest beta release (September 2020), but still get only 5V input to TS80. With wall PS it easily feeds by 12V input and heating blazing fast))

Ralim, 2020-11-01:
@salavatkhafizov Bugger
Any chance your Saleae can do analog capture of the D+ and D- pins as it attempts to negotiate the QC ? Would be curious to see if anything odd is showing up there

salavatkhafizov, 2020-11-02:
@Ralim I will try, but do not know when (( I do not have logic analyser where I am at the moment.

Ralim, 2020-11-03:
No issue when :smile: Just would like to figure this one out :)

salavatkhafizov, 2021-03-26:
@Ralim I've got another power bank (Xiaomi Wireless 10000 mAh), latest firmware works without an issue (up to 12 V ps).
Lets consider this issue closed)))

closed 2021-03-26 by salavatkhafizov