Ralim/IronOS #456: TS80: Temperature shown is way lower than actual temperature of the tip

salavatkhafizov, 2019-03-13:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ x ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
Whatever temperature setting on the TS80 iron is incorrect on any firmware, about 80 C lower than actual temperature on the tip (D25).

* **What is the expected behavior?**
Set temperature should be within 10 C from the actual one.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.  Make a temperature calibration
2.  Set temp to 180 C, actual temperature is about 250C, the same with set temperatures at 250C, 300C, etc
3. Measurement was done by 2 different instruments.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
Should work more or less accurately, major flaw at the moment.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x (any)
  - PCB Version: (1/2) (not sure)
  - Power Supply (Voltage and Current Rating) :
9V 2A, 12V 1.5 A (supplied with the iron)

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-03-13:
Can you confirm if you have done any calibrations?
If you have done any, can you please check before and after?

salavatkhafizov, 2019-03-14:
Yes, I tried many calibrations (leaving iron alone for few hours, immediate calibration without heating the handle).

Both on stable firmware, with calibration (I can confirm that in menu it is say User Tip o something like that after calibration) or without (doing factory reset) the same results - set 150 C, actual temp about 280 C.

Ralim, 2019-03-14:
Could you please list out the readings from the debug menu along with your room ambient temperature level please.

salavatkhafizov, 2019-03-14:
Tip Model - User Tuned
V 2.06 TS80
Dec 16 2018
Heap: 002536
HWMG:000906
HWMP:000452
HWMM:000414
Time:000314
Move:000325
RTip:001395
Tip:000003
Vin:09.9
THan:033
Model: 2
QCV:090
Tr:004285

Ambient temp is about 28-29 C.

Thanks!

salavatkhafizov, 2019-03-14:
I double checked those on 5V input - they are more or less (almost) the same.

salavatkhafizov, 2019-03-14:
Measured D25 tip - it reads 4.45 Ohms.

salavatkhafizov, 2019-03-15:
@Ralim  Please review for future references if needed and close it.

1) Considering issue closed, since it looks like a hardware problem and sent back to seller for exchange.
2) For those who dare (or have) fix themselfes:
- similar issue was noticed by TS100 users,
- the fix was to reflow part of the iron, with 2 chips.
- on TS80 chips, responsible for temperature measurements look like this:
[image]
- those 2 chips most probably are:
***T6G*** is likely TMP36 (
https://www.analog.com/media/en/technical-documentation/data-sheets/TMP35_36_37.pdf)
***S06HF is likely SGM8511XN (
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=2ahUKEwjjrM7F54ThAhWJepoKHQl2B70QFjAAegQIMhAC&url=http%3A%2F%2Fwww.chinesechip.com%2FchipFile%2F2015-07%2FSGM8551XN-18456-0.pdf&usg=AOvVaw1FZn6b_M7K5zZT0W2SIUJW)
in appropriate cases.
Those are available for replacement, if that is required.

Ralim, 2019-03-19:
@salavatkhafizov Thank you for confirming this.
It is of note that in your above debug information, the handle temperature was reading as around 3 deg c which does suggest an issue with the handle ambient temperature measurement.

T6G is indeed the handle temperature sensor. Out of curiosity, I wonder if the trace under it is heating up under load and causing thermal issues with the sensor.

I'm sorry that it was a hardware issue :(

closed 2019-03-19 by Ralim