Ralim/IronOS #457: Investigating enabling Weller RT tips

hedgeberg, 2019-03-14:
* **I'm submitting a ...**
  - [ ] Bug report
  - [x ] Feature request
  - [ ] Translation

* **What is the current behavior?**
Unknown -- google seems useless on this one

* **What is the expected behavior?**
Ideally, make Weller RT tips compatible after some converter is slapped in between controller's 3.5 mm jack?

* **What is the motivation / use case for changing the behavior?**
Enable use of expanded set of compatible tips!

* **What are you running:**
TS-80 FW 2.06

So, here's the question: would it be possible to enable the Weller RT compatible tips to be used if one were to remove the cover and extend the input barrel of the TS-80 with a 3D-printed cylinder containing a plug-to-socket (male-to-female) adapter internally? if so, what would be necessary? Are there differences in pinouts between the weller and minidso tips that would need to be bridged in the adapter? Would some kind of impedance matching network be necessary? This is all something I'm willing to experiment with/play with myself to get it working, mostly just curious if this has 1. been attempted, 2. is possible. Theoretically, if a small adapter is thrown in between the different tip sets, theres no reason this couldn't be done, is there?

(Also, hey Ben! How's life in the twitter hw hacking community? haven't missed anything too big, have I?)

Ralim, 2019-03-14:
Hey!
Now that's a familiar name to see around these parts!

So, as far as it's understood, the pinouts most certainly are not compatible by default. Miniware uses a thermocouple that is part of the heating element (like hallo) where as the Weller tips use an extra 3rd connection for the thermal sensor.

To make this work would require either a mood board inline, or rework of the main board to remap pinouts.

I can get back to you later on with Pinot details for the ts80.

It would be neater to mode the board as then you don't need to turn off the pwm while taking a temp measurement.

I have spent a few hours talking to miniware about this and USB pd, but seems like I can't get either on the roadmap anytime soon (that they will tell me about).

Have started toying with trying to make an alternative main PCB for inside the iron to support both styles of tips and USB pd as well as QC. Though lately I've barely had any free time to work on it :(

But basically: definitely needs hardware mods for the pinout, the ts80 tips are 4.45 ohms, would most likely need a new adapter to screw on the end to hold them properly.

I don't think you have missed a huge amount that I've noticed out there, I'm mostly enjoying the photos and layout work of gregdavill. And watching projects happen, haven't seen any huge big rev engineering efforts lately that come to mind so far.

hedgeberg, 2019-03-14:
Aw that's true, I'm missing greg's stuff... c'est la vie, unfortunately. 

Anyway, to make sure I understand, miniware is using a 2-port system, with power and impedance on the same pin, and the 3rd pin on their audio barrel connection is unused? And weller uses 1 pin for sense and 1 pin for drive? admittedly, sounds like a frustrating fix, but it should actually be doable with an inline board, without needing to mod the board itself. I need to look at how the weller iron's sense behavior is done (or how I could get away with reading that inline), but its a /lot/ easier to go from split sense and drive to a merged sense/drive than it is to go the opposite direction, so that's good. I don't know that I could make something work for using the ts80 tips with the weller control unit, but i think, if my understanding is correct, i do have what it takes to make the weller tips work with the ts80. May require some tricky fixes (like a digitally controlled impedance, which i'd need to figure out how to power parasitically or something), but it /should/ be doable. I'll look at rigging up some kind of method for investigating this w/ an oscilloscope at some point (maybe a good streaming project actually?) in the next couple... months? Let me know if you make progress on the alternate board idea, don't want to duplicate work, and I'll slap any ideas/designs i come up with here.

Ralim, 2019-03-18:
Miniware use a 2 port system with the thermocouple inline, which means that the thermocouple passes the heating element current, which is what makes an inline adapter a bit of a pain.

Currently this software measures the thermocouple during the off state of the driver PWM signal.

I'm more inclined to make a replacement main pcb for driving the RT tips, and fix my biggest issue with the TS80, the lack of PD support. (I know PD is a trashfire in terms of a protocol, but almost everything I own supports it).

But in terms of an adapter, i think its viable to modify the feedback sense connection to the unused connection in the normal Miniware tips (Need to check if this is an open contact inside the tip) and then that will allow for the temperature to be measured without needing to pass the main current through the measurement tip. Might also be viable to use a mosfet to short out the sensing path during current drive or something as well, though not sure, and also dont overly want to risk breaking at $70 tip to find out :D

The other thing is that i believe the weller tips are lower resistance, so not sure how well the ts80 will take the higher current spikes (and your QC source). The software does have the capability to measure the tip resitance, so could unlock the software measuring that low and so it can ask for < 9V to keep below the 2A hardware "Limit" as a work around fairly easily.

If you do look at this and do a stream, be sure to post here so i know about it :) 
More than happy to work with you on this too, though not sure how much i can contribute, fairly sure your knowledge goes far more in depth than mine in most of this :grin:

Ralim, 2019-08-27:
Slight update on this,
A kind fan has sent me a Weller RT tip in the mail that im expecting in ~ 2 weeks.
Hopefully after they arrive I'll be able to have a more in depth answer on this and a possible timeline estimate for support (and any hardware changes required).

joric, 2019-12-06:
They are hardware-incompatible.

[image]

Ralim, 2019-12-10:
@joric 
I have been doing some testing.
Apart from the barrel sizes (8.3 vs 9mm), they are compatible with some hardware mods (so far just an extra resistor). They also have different temperature curves that I need to profile first.

But it does look like its somewhat compatible , but does require a touch of legwork

mgrapiniak, 2019-12-17:
@Ralim It happens to be the case that I have both Weller tip and TS-80! Willing to use this power for testing beta releases if needed ;)

joric, 2019-12-17:
@mgrapiniak is it possible to insert Weller RT in TS80? I was under impression the extra 0.6mm prevent inserting it into the connector even with the sleeve removed.

mgrapiniak, 2019-12-17:
> @mgrapiniak is it possible to insert Weller RT in TS80? I was under impression the extra 0.6mm prevent inserting it into the connector even with the sleeve removed.

@joric: That is true. It is not possible to slide the tip into TS-80 without physically removing that threaded thingy :( (not sure how to call it in English.. sorry :P )

joric, 2019-12-19:
@mgrapiniak sorry still unclear. You mean without sawing the sleeve base off? This sleeve base that can't be unscrewed, it should be either removed or drilled up to 9mm or the Weller tip won't make contact.

[image]

Ralim, 2019-12-21:
There are three issues with using Weller tips :

1. They dont fit, and will require a mechanical adapter of some kind
2. They have seperated out the temp sense and heater element.
3. They have a different response for the temp sensor.

2. Is solvable using a resistor I believe
3. Is _just software_

1. Is the harder of the triplet, but is most likely solvable with an alternative case for the TS80, or some kind of mechanical adapter, or by modifying the weller tip.

mgrapiniak, 2019-12-21:
> @mgrapiniak sorry still unclear. You mean without sawing the sleeve base off? T
> 
> [image]

Yes. That's what I meant.

mgrapiniak, 2019-12-21:
> There are three issues with using Weller tips :
> 
> 1. They dont fit, and will require a mechanical adapter of some kind
> 2. They have seperated out the temp sense and heater element.
> 3. They have a different response for the temp sensor.
I find this link to be quite useful in terms of alternate solution as well as some nice temperature graphs (if you haven't seen it already) http://kair.us/projects/weller/index.html
> 4. Is solvable using a resistor I believe
> 5. Is _just software_
> 6. Is the harder of the triplet, but is most likely solvable with an alternative case for the TS80, or some kind of mechanical adapter, or by modifying the weller tip.

I would say not an option in terms of weller tip modifications :)

whitehoose, 2019-12-21:
Let me start (yes again!) by saying I completely get it.  I've been modding stuff for years.  Most recently I bought a brass shepherd's whistle (long story - but they are an instrument - I wanted to use one - but the range isn't ideal).  I ended with something similar using shapes cut from an old credit card - it sounds perfect.    If I was to cost it at my usual rates - 3 months work and materials - it's cost £40,000 - to me the satisfaction is priceless - and believe me I do see the irony but I'd do it again in an instant.  It was fun while it lasted.. 

In the UK we have a comedy series called Only Fools and Horses (a bit of a classic). In one episode 
https://www.youtube.com/watch?v=56yN2zHtofM the legend of Trigger's broom was born the concept is shown in boardrooms and at staff project management training events throughout UK as an example of what can happen for anyone considering starting a project.....
For broom sub TS80 ... or TS100 - work the rest out yourselves.

Have a good christmas!

joric, 2019-12-24:
> Have a good christmas!

You too!
* https://github.com/joric/ts80player

**[TS80 Soldering Iron + Audio Player (Merry Xmas!)](https://youtu.be/aose7zWV1fM)**

[image]

i2, 2020-01-19:
> I ended with something similar using shapes cut from an old credit card - it sounds perfect. If I was to cost it at my usual rates - 3 months work and materials - it's cost £40,000 - to me the [...]

I'd like to see a picture of final product :)

whitehoose, 2020-01-19:
You can see the striped credit card superglued.  Sounds surprisingly sweet.
The other made from 3 layers of 1.5mm pvc.  I use 4mm brass belt rivets so I can trim and tweak  This one is finished and I'll trim the rivets to improve the lines.

[image]

ammoniak, 2020-01-31:
I'm going to try to build an adapter that fits between the TS80 and the weller tip... Is there an easy way to change the temperature profile in the software without recompiling everything?

Ralim, 2020-01-31:
You will have to compile the software with the temperature profile of uV<-> deg C

Compiling is fairly easy these days using either the ide or command line / docker etc

DurandA, 2020-01-31:
> I'm more inclined to make a replacement main pcb for driving the RT tips, and fix my biggest issue with the TS80, the lack of PD support. (I know PD is a trashfire in terms of a protocol, but almost everything I own supports it).

@Ralim Are you aware of https://github.com/Jan--Henrik/Otter-Iron?

Ralim, 2020-01-31:
Indeed I am aware of it 😅 I have one even  except that I need to replace the screen on it before I can start using it

i2, 2020-02-07:
@Ralim where did you get the Otter-Iron PCB from?

discip, 2022-04-23:
@hedgeberg
Please close this issue, since this is not worth the effort as stated by @Ralim [here](https://github.com/Ralim/IronOS/issues/457#issuecomment-568169322). 😊

thanks

hedgeberg, 2022-04-25:
Ah, haven't been paying great attention to my gh notifs. noted. closing. sorry this has been open so long!

closed 2022-04-25 by hedgeberg

Ralim, 2022-04-26:
No problem, sorry I never really had time to dig into this either :(

hedgeberg, 2022-04-26:
Such is life in this sphere, unfortunately. Too many projects, too little time.