Ralim/IronOS #458: TS80 voltage reading is off by >2V

l0ud, 2019-03-20:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
 Report bug

* **What is the current behavior?**
I cannot get reliable voltage reading on my TS80. Looks like readings (at least in standby) are random in QC mode and calibration only compensate error to the first change of voltage.
Problem does not occur on stock firmware - voltage difference is about 1V when heating, but almost not visible after reaching requested temperature (thus turning off heater).

* **What is the expected behavior?**
I expect to read correct voltage all the time, or at least in standby mode, when there is almost no voltage drop.

***Steps to reproduce:***

1. Connect TS80 to supplied USB power supply via USB power tester and compare read voltages on TS80 and USB tester. [I was reading 8V on TS80 when USB tester measured 9,6V]
2. Callibrate voltage to TS80 to compensate difference in reading.
3. Turn on soldering mode, see voltage fluctuation. Go back to menu.
4. Read difference between USB tester voltage and TS80 once again. [I managed to read 10,1V when the charger was giving ~6,4V]

* **What is the motivation / use case for changing the behavior?**

I would like to see correct voltage on my iron. I suspect temperature regulation is inaccurate when voltage is read incorrectly.

* **What are you running:**

Firmware Version: 2.06-RC4
PCB Version: (1/2) (not sure, it's displaying model 2)
Power Supply (Voltage and Current Rating) :
9V 2A, 12V 1.5A "DN.OX" (provided with TS80).

Ralim, 2019-03-21:
That's.. very odd 

Can you confirm the file you flashed to your ts80?
I'll try and see if I can reproduce the issue.

There isn't much filtering on the input voltage, though it's normally rather stable.

l0ud, 2019-03-21:
I'm using exactly this version (Polish language):
https://github.com/Ralim/ts100/releases/download/v2.06-RC4/TS80_PL.hex

At home, I'll verify the checksum of the downloaded file (sadly I cannot verify checksum on iron). It doesn't look that the measurement is not stable, at least in standby. Once in that mode, it's just reading consistent, wrong value.
Of course, this observation is wrong if your firmware is not refreshing voltage readings in modes other than soldering.

Ralim, 2019-03-21:
Checksum validation on the device is a pain (no easy correlation between hex and whats in flash).
Can I also ask you to test the English version as well to narrow down the possible causes?

Voltage readings are refreshed at around 100Hz.

l0ud, 2019-03-21:
I've tested English firmware and it has the same problem.
After connecting to charger and factory reset, it starts from 5V and slowly increasing (it takes 6sec) to 7V, while charger is putting 9V constantly. After turning on soldering mode and going back, the difference can be so big, it is outside the calibration range...

l0ud, 2019-03-21:
I just disassembled my unit to verify its PCB version: 1.7. It's exactly the same model you can see on eevblog review: https://www.youtube.com/watch?v=_Z9es-D9_8g
I visually checked values on all resistors and they are all equal to what I can see on that video. Then measured voltage directly on type-c socket thinking that my usb tester may be wrong, but no, it's exactly 9V. I can probe more if you think it would help with debugging that problem.

denesstreambright, 2019-04-12:
I can second this, for mine it sais 7.3 - 7.6V for the factory charger in info mode (The original v. 1.06 was OK, 9 +- 0.2V)
Also when in detailed soldering mode, it displays a different voltage reading, and it constantly changing (running from 8.0 to 9.9V and back constantly in a ~3 second cycle)

Ralim, 2019-08-27:
I have removed the QC auto-adjust that was in RC4, in RC5. 
This should solve most of these issues.

Re-open or comment if this is not solved for you on the newer releases

closed 2019-08-27 by Ralim

l0ud, 2019-09-08:
It's still kind of weird.
After boot it shows 5V and then slowly climb to 7V. USB voltage meter says it gets 9V.
After turning on soldering mode, voltage reading climbs up to 9V and stays here (even when I turn soldering mode off), real voltage is still solid 9V. Shouldn't it negotiate 12V? This is QC 3.0 powerbank capable of outputting 12V.

whitehoose, 2019-09-08:
The ts80 uses a 4.5 ohm tip and is rated at 18w, (appprox 2A) so @9v looks pretty good.
If you ever really achieve 12v your 18w iron will pull 32w and I'd guess go into meltdown mode.

You have an Iron that incorporates an approximate voltage indicator
on the other hand your usb AVO is a dedicated measuring instrument that (as far as I'm aware) doesn't solder.  If it looks like an AVO, Smells like an AVO - and acts like an AVO then it's reasonable to think it's probably going to be more reliable than your soldering iron.

Ralim, 2019-09-08:
@l0ud 
The slow ramp up is completely normal  that view has a very heavy filter on it that takes time to react to the QC step change.

It will only negotiate for 12V if you have the iron in the higher wattage mode in the menu. By default it only goes for 9V as this is safest.

l0ud, 2019-09-09:
Well, I can wait like 5 minutes and it's still showing 7V - it doesn't looks right. After going into soldering mode it's immediately jumps to 9V.
I have power set to 12V (it's showing PWSC12V) but it's still negotiating only 9V. I checked on wired charger and it's still only 9V.

Ralim, 2019-09-09:
It should settle up to 9V, so something is off there...

Does your power source support QC2 or QC3? Does it support QC2 voltage levels or only QC2 9V plus stepping?

There are somant different implementations of QC. In the least release I went to negotiation for 12V using the QC2 levels rather than stepping ask all QC3 devices are meant to support this, but a few people are running into issues as it seems quite a few dokt implement this.

ironman123-eng, 2019-10-17:
> That's.. very odd
> 
> Can you confirm the file you flashed to your ts80?
> I'll try and see if I can reproduce the issue.
> 
> There isn't much filtering on the input voltage, though it's normally rather stable.

Hi ralim I’m getting a error when I copy the file over

Ralim, 2019-10-21:
@ironman123-eng 
Its best you start your own issue rather than stepping on another one