Ralim/IronOS #459: Temperature display does not match real tip temperature

blitzcode, 2019-03-22:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**

TS100 with a D24 tip and tip set to D24, temperature calibrated, set to 300C, measures ~280C with thermocouple and two different tip thermometers.

* **What is the expected behavior?**

300C.

* **What are you running:**

v2.06-RC4, 19V PSU.

* **Other information** 

Is the temperature supposed to be accurate now? I tried all kinds of different releases of this firmware over time, temperature readout was never really accurate. Is this just an unfixable, fundamental issue or is this a bug / something I'm doing wrong?

Kubuxu, 2019-04-30:
I'm working on a version of the software that uses a more precise thermocouple model.

With it, the TS100 is able to measure the temperature of the tip within about ~2C under the condition that tip is isolated and allowed to reach temperature of the core.

I've observed about 20C of difference between the tip surface and TS100 reading in open air. This delta is due to the physical placement of the thermocouple within the core. With precise temperature readings and good physical model, I hope to be able to compensate for that in future.

Ralim, 2019-05-01:
@Kubuxu
If you implement a better thermocouple model, please raise a pull request, more than happy to bring in improved models and logic! 

Big stalling part of most of the temperature problems is lack of time to perform testing, modelling and implementation.

Kubuxu, 2019-05-01:
It is still WIP, I will create PR once I clean it up and make sure it works (there are still some bugs I have to get rid of).

Booroondook1, 2019-06-12:
Tell me, what should be the temperature in the room to calibrate?

Ralim, 2019-12-29:
Hopefully the latest release has addressed the temperature calibration to be close for most people (keeping in mind that for soldering, you're going to get like +-5 deg C anyway).

> Tell me, what should be the temperature in the room to calibrate?
Doesn't matter, only matters that the handle and the tip are the _same_.

closed 2019-12-29 by Ralim