Ralim/IronOS #460: No logo on TS100 v2.06 & DFU 3.45

santiaguin13, 2019-03-26:
Hello 

I have my TS100 on v2.06 with DFU 3.45 and I have also updated it to the last spanish firmware.
I have tried two different methods for changing the logo: using the "TS100 logo editor" and tried to copy a .bmp logo on it but I couldn't do it. The image is 96x16 and using the "TS100 logo editor" the .hex file changes to .rdy succesfully but the logo doesn't appear either.

Any idea of what's happening here?
Thanks.

compgeek35610, 2019-04-04:
This could be the same issue I had with setting a logo on the TS-80 using the logo editor?

I updated it again to v2.06 RC4 then used the Python logo tool to create the hex file. Not sure why the exe didn’t work, but Python did the trick.

santiaguin13, 2019-05-05:
Thanks i will try with python but sorry i'm new in programming and i don't know how to use the python file. I've read it but i don't understand most of it. If there's a video or something explaining how to use it i will be very gratreful if you help me.

maciejmatczak, 2019-05-10:
Hi guys, I tried following:
- on windows: .exe & .py creators on https://github.com/Ralim/ts100/blob/master/Bootup%20Logo/Logos/001_TS100.png, both failed
- the same image and .py creator (but used on linux) generated working .hex
- actually, directly downloaded .hex file from github also failed I believe...
- on windows, image drawn in gimp, saved as .bmp. and .exe creator worked fine

If something needs to be double checked please tag me. GIMP + .bmp on Windows got me working custom logo.

d-freshmaker, 2019-05-16:
I am having the same issue.  I have tried several different logos, but everything always ends with a .err extension.  My initial logo was created in MSPaint and converted via the Logo Editor.  I have also tried a converted .bmp created in Gimp (win), a converted downloaded .bmp image, and a .hex logo file that worked for a Youtuber I follow.

santiaguin13, 2019-05-16:
I don't even receive back a .err file. It all seems to be okay but the logo doesn't appear.

d-freshmaker, 2019-05-17:
For what it's worth, today I also tried generating the .hex file in Linux, using wine and the Logo Editor.  I copied the .hex file to the iron also using Linux.  I was still given a .err file after the iron reconnected.

oiitsjamesmate, 2019-05-19:
Also have this problem. I flashed the TS100 with a completely white logo and now whenever I upload a new one, it says RDY but jumps straight to the soldering menu with no iron. Tried re-flashing firmware (Ralim and Mini) as well as formatting but it didn't complete.

Ralim, 2019-05-29:
Please try using the python generator tool, as it's far more tested than the c# app and it handles images better if they are not ideal.

Also please test with English firmware first to rule out language related issues.:)

d-freshmaker, 2019-05-29:
I just tried converting some logos in Linux using the python generator tool, and they still produced .err files when copied to the TS100.  I tried downloading the 001_TS100.hex logo from ts100/Bootup Logo/Logos, and it produced a .err file as well.

I just purchased this TS100 a few weeks ago, and it came with DFU version 3.45.  Is that the problem?  Is there a way to downgrade to 3.43?

Thanks!

arminth, 2019-06-08:
Same with me. Brand new TS100 arrived yesterday. DFU 3.45
No chance to get Logo working. Always error. Both with python produced logo-file and the one provided here. 
I cant solder without a logo!! ;)
If you could check, would be nice!
Other than that awesome job!

Cheers
Armin

Ralim, 2019-06-09:
This appears to be a change on DFU 3.45 in that it has become significantly more picky about the HEX files that are sent to it. At the moment I'm not exactly sure what it is looking for, for it to be happy with the hex file.

shortblack, 2019-07-08:
@Ralim I'm seeing the same issue on my TS100, running DFU 3.45 only the stock firmware works for me. Would there be any useful debugging info that I could try to provide?

My background isn't in this kind of programming, but willing to try figuring it out if it would help.

timecop97, 2019-08-14:
@arminth @Ralim Hey all! Just been trying to get the TS100 to accept a boot logo. I was getting the .ERR extension, and i noticed my DFU was V3.45, and that it accepted the logo hex for the TS80. How was that TS80 logo generated? located at ts100/Bootup Logo/Logos/001_TS80.hex

Ralim, 2019-08-15:
As far as I'm aware, every logo in the repo was generated using the python script :)

PR #491 (2019-08-15): changed intel Hex gen to use unix line endings to fix #460
Please try and fill out this template where possible, not all fields are required and can be removed.

* **Please check if the PR fulfills these requirements**
- [X] The commit message make sense
- [X] The changes have been tested locally
- [ ] New features have been documented in the Wiki
- [X] I'm willing to maintain this in the future (Totally Optional)

* **What kind of change does this PR introduce?** 
This fixes #460 where logo .hex files are rejected and produce a .ERR file on the DFU version 3.45

* **What is the current behavi [... cut]
commit a18cca66 (2019-08-15): Merge pull request #491 from timecop97/master

changed intel Hex gen to use unix line endings to fix #460
closed 2019-08-15 by Ralim