Ralim/IronOS #462: TS80 Power source doesn't work.

Gregory1981, 2019-04-12:
I am submiting bug report.
Whenever I chose in menu Power source 18W or 24W it takes whole 30W or more in peak what cause my powerbank to cut off as it is overloaded.

Is the option 18W or 24W limits power taken from source?
If yes, the it is not working on my powerbank (Soshin e3s) or mains USB QC charger. I have checked and for both it takes maximum what power source can supply what is 30 or more watts.
Is that possible to fix it or I do not understand properly functioning of this option?
If I am wrong and that option is for something else, please delete my request.
You are doing great job Ben.
Thank you.

Firmware version is the latest one 2.06-RC4.
I do not know how to verify PCB version.
Power supply is wall charger USB 3.0 QC and PowerBank Soshine e3s.

DjordjeMandic, 2019-05-19:
18W Original adapter 9V 2A - 12V 1.5A
[image]
[image]

24W Original adapter 9V 2A - 12V 1.5A
[image]
[image]

Now here's the problem: When switching back to 18W from 24W voltage does not drop to 9V and it stays at 12V and the power is still 28.6w, you need to do power cycle.

On samsung eb-p1100c (9V 1.67A) in 18W mode voltage is 9.3V and power is 16.7W (also sometimes it's higher than when using the original adapter). In 24W mode voltage is oscillating at around 9V - 9.3V and power is oscillating at around 16.5-17.5W which is normal because of the overloading of the powerbank.

Main problem here is why it goes over 24W when set to 24W and does not go back down to 18W when set to 18W without power restart and why it does not pull 18W at 18W setting instead of 16.2W?

Still how the fuk does it manage to pull more than 18-24W? 

@Ralim can you confirm this?

whitehoose, 2019-05-19:
Standard processor operation - once the mode is set it won't change without a reboot even linux has to be rebooted after a hardware state change.
You also have a reset option - some changes require that too.  It's good practice to get into the habit.
If you set out to read EVERY thread on both ts100 and ts80 you'll find some answers.
Also don't forget Ralim isn't paid for his work he does it for pleasure - don't kill the goose mate.

Ralim, 2019-05-21:
@whitehoose  Is correct here in that the power limits only take effect when the unit is restarted as they adjust the negotiated voltage from the QC source.

All power measurements shown on the unit are a "moderately educated guess" formed by measuring the tip resistance, estimated PWM duty cycle, and the measured supply voltage. Do not trust these when having issues with power supplies turning off or not, as they are probably +- a significant amount depending on the tolerance of your iron. They are there as estimations as a more useful indicator of "load" than anything.

The way 18W and 24W mode work is at the moment is in the voltage negotiation phase rather than cycle by cycle PID limiting (thats coming, but sorta waiting on a neat looking patch from @Kubuxu  before digging into it again). 

At the moment the flow for 24W is to request 12V and let the iron go. On almost all TS80 tips, this works out to a "peak" possible draw of around 26W, but it is only at that limit for around 1 second before it starts to drop off after the first ~80 deg of heating.

For the 18W mode, the tip resistance is measured, and then it back calculates the best QC voltage to request to ensure that 100% duty cycle is approximately 18W (rounding up the voltage request).

> Still how the fuk does it manage to pull more than 18-24W?

Being rude only makes the answer take longer to get written :smile: , this is by far my favourite project to work on when i have time, and I'm always happy to answer questions and take feedback at any time.

That said, if you could please measure the power draw with something accurate (inline power meters are good) that can help diagnose why your supply is shutting down. Please note that the `eb-p1100c` is not rated for 24W mode, and it may not allow full power at 12V output voltage either.

DjordjeMandic, 2019-05-22:
@Ralim I tough that power is measured via current. My bad. Thanks for letting me know!

Ralim, 2019-05-23:
I wish there was a current measurement however there isnt, so it's all speculation :)

whitehoose, 2019-05-23:
@DjordjeMandic see here
https://www.rapidtables.com/calc/electric/power-calculator.html#dc

Anyone with a soldering iron and an urge to build, mod or hack (why else would you need a soldering iron?) should start by getting ohms law and basic power calculations fixed in their head.
They are the foundation to every decision you take with the magic pixies.  Look round the room, what fuse values should you expect for each device?  Spotting the high current might save your life.

A basic power meter (get a USB and mains version) will help you build a picture and get a basic idea of what to expect things to do.  Easier than a meter to start with - but start saving because a meter is a must have too.

The magic smoke is constantly trying to get out, and the pixies use current, voltage and resistance to keep it in line.  An heating element or LED is one of the classic "small changes make big (and often violent) results" - 1/2 an ohm or an extra volt can make .... 
Use the calc to calculate what they can make.

You really do need to have a feel for how they work together before you can judge if it's a fault or f/w that's causing your grief.

Mpic, 2019-08-06:
> measuring the tip resistance
I don't understand how you measure the tip resistance without measuring the current that passes through it. As far as I know, "measuring resistance" must be done by measuring voltage while knowing how much current passes through...

whitehoose, 2019-08-06:
Mpic. - It's easy to get a ballpark "cold" resistance reading.
Switch AVO to ohms, pull the clip to tip and ring - read numbers .... easy really!

In reality, under load the readings will change with rise in temperature and drop in resistance, depends how exact you want to be with the numbers, my gut tells me it's going to be a relatively small swing- the ts80 is rated at 18w - if resistance halves @9v that's going to double the power and I imagine 36w will nuke the iron.  Normally small values mean small changes. 

If you want a "real world" working reading you need to breadboard a short 3.5mm jack m-f extender (I'd use vero board) .... terminate with tip and ring connection "broken" by 2 pairs of bridging pins.
Plug male of extender into iron - plug bit into fem of extender.  Remove 1 bridge link.

You need avo 2 meters - 1 on volts across tip and ring bridge pins.
The other on current  bridging EITHER tip or ring continuity.
Switch meter to appropriate current range.

on a ts80 during heating - the tip is about 4.5 ohms cold,  it's going to pull about 9v while heating - so to be safe I=V/R you need to be on a <5A scale.
take readings .... crunch the numbers. R=V/I

Mpic, 2019-08-07:
Whitehoose - You mean the software doesn't measure the tip resistance, it takes a ballpark value measured with a DMM while cold ? 
I have yet to get a look at the code.
I assumed there was at least a shunt resistor in series with the tip to measure the current, which would enable precise power readings.

Ralim, 2019-08-07:
No shunt resistor, everything is sort of guesswork :) 
The TS80 has a oullup that lets us measure the tip resistance, but it's only as accurate as measuring it over it step range.

So both code paths have rough resistances and it's all guesswork from there :)

whitehoose, 2019-08-07:
> Whitehoose - You mean the software doesn't measure the tip resistance, it takes a ballpark value measured with a DMM while cold ?
> I have yet to get a look at the code.
> I assumed there was at least a [...]

MMMMM not really ... even a simple AVO is quite complex and it's built to do the job.
The TS range are built to get hot within certain parameters and to maintain a temp to +/- some other tolerance - solder isn't really that fussy.
If you were working in aerospace or deep space - you wouldn't be using a cheap ts iron.

By adding a thermocouple you detect when your iron is changing temperature and you can jiggle the pixies to raise, lower or maintain that chosen temp.  But after 50+years of melting lead I still couldn't look at a job and tell you (to 0.1 of a deg C) what temp to set your iron to for a good result.

It's not a requirement for good soldering.  You look at the job and estimate the amount of metal you need to heat up to make those molecules meld, you dab the site with solder and watch for it softening then look at how it melts and flows.  Every job is different and if required you jiggle the settings accordingly.  Unless you're working at the extremes mostly within a few degrees you get away with letting it be.  It's only quite recently changing temp has been an option at all. 

If you bracket a temp quite widely (10c) you eliminate (or at least reduce) the effects of variance.  It's never going to be an exact thing, esp outside a lab. 
Short of xrays or a saw - you rely on your experience to guesstimate if it's going to fly.  No iron will ever tell you that.  Too many variables - you don't want to rely on numbers - Trust your eyes.

Whatever the chips CAN do ... I Just don't see the point in expecting (or paying for them) to emulate a fluke meter.  I find my ideal temperature for each job by observation.
... Once it's set I expect the iron to maintain it reasonably accurately.  It may shock you to know - I don't actually care what that temperature is apart from it being the one I've settled on.  I do want it to achieve that temp +/- 10c ish everytime I turn on.  But that's as far as I need.

If it will compensate between thin tracks by ushing more pixies in for a thick bus that's handy - but there will be limits for what is credible and inevitably the tip contact time for any temp will vary with mass  (and it's going to be the temp I want ... not any particular number).

I don't expect (or want) an iron to do my job, just to maintain a ballpark temp over a reasonable range of materials and mass densities. 

I'll tell you my iron is set at somewhere round 320c ... ish ... for the jobs I'm doing now .... 
I do know that I'll never set my iron to >190c ... and that I know 300c is the ballpark spec minimum temp for my favourite brand's 60/40 - I've never needed 400c - I doubt I ever will - the solder I use doesn't like to be above 380c but the components aren't too fussy (some mosfets are very critical). 
The rest is guesswork and I don't know if that's going to apply tomorrow.

When I was a boy a well designed iron was one that melted solder and not it's own tip - end of spec.  Thermal mass and drain were considered to account for the cooling effects of expected materials, but if it didn't do the job you required - it was because you'd not bought the correct iron.... tech support would tell you you needed to rethink your choice.  Part of the training was appreciating what wattages did what.  I still carry a choice and I own probably <30 between 12 and 250w

Mpic, 2019-08-08:
Thanks for sharing your experience.

I don't think a precise temp is really needed either. But since only a shunt and a voltage measure is needed to get a precise power reading, I regret Miniware hasn't made the choice to include one.
And since the software provides seemingly precise power values, it's really important to advertise that these aren't real measurements but ballpark values.

It wouldn't be as important if the voltage sources used for the TS-80 weren't that finnicky...

whitehoose, 2019-08-08:
I use my ts80 pretty regularly connected to a number of sources, in fact it's become a bit of a game to see just what will and won't suffice.  I settled on the AOKcoda qc3 USB charger as my supply conditioner (it's cheap enough to use as a consumable and always talks to the TS80 very nicely).  Connect that to anything you want, I use a buck boost converter to push the envelope and I can power the iron with anything from a button cell up to to a 35v source.  But that's just a game for giggles.
For real work, once you've got qc3 chatting to the chip - and I agree, with a seemingly infinite range of "qc3" protocols used by powersources the rest is a lottery with many pitfalls.
Achieving a digital voltage reading on the other hand isn't quite the walk in the park you make it sound, nor is translating that into power readings/temperature and once you have it what does it achieve? ... the adjust you need is basically up or down.  Once qc3 is chatting - it sorts the rest without any need for user intervention.
As long as the solder flows who cares what the numbers are?  My only concern is not to destroy the iron, and the AOKcoda is built specifically for that purpose.  The rest is maybe nice to have, but it's nice, not vital.

If you're doing specialist builds - you buy specialist equipment to fulfil your specific needs - In my time I've spent ridiculous money in order to fulfil very rewarding contracts.  What you want is doable .. but not in this price bracket.  You want cream you need to add a couple of "0"s and look at better kit.  The hardcore pros mostly use far simpler hardware built to more exact tolerances.  My real work is still done with a hakko - I got a TS to see if it brought anything better to the table than my 20 year old portable RS NiCad rig.... I avoided the 100 because it seemed closer to being a real iron, the 80 seemed more interesting - I live in hope one day it will be. 

Ralim and the others are doing a good job at improving an iron that's arguably strayed from the path a little.  KISS is always the best advice for real world use - projects fail when people lose sight of that objective.  Tetris mode for the ts100 just illustrates that the resources provided aren't necessary for what's needed to melt solder  ...

Ralim, 2019-08-27:
Okay, so I would love to add this sort of a feature to the iron.
Now that the big font cleanup is done and there is some room back in the flash on the device I'm looking into this (ever so slowly).

I'm thinking of changing the current menu to have an option for the voltage that is negotiated for (9V, 12V QC2, 12V QC3, No QC...) and then having a separate option to limit the PID output to try and limit the power drawn.

Would this work for you ?

Gregory1981, 2019-09-20:
Yes. Lets try out this new option :)
Thank you Ben for your hard work.

Nazosan, 2020-01-18:
I'm not sure if this is the right place to put this, but I saw the description for the latest firmware and its much more controlled power limit and was excited to try it since finding a decent power supply that can handle the power this wants without burning itself and possibly my expensive TS80 out with it is kind of hard to do.  This new firmware seems have a great idea for how to handle this, but with my TS80 it seems to be limiting it quite a bit below whatever I set it to and it heats up _VERY_ slowly (less than 2C per second) and struggles to maintain temperature.  For example, if I set it to 18W at 9V it will limit to 15.8W instead.  If I set the voltage to 12V it goes up to 17.0W instead of 15.8W, but still heats up incredibly slowly.  I realize that's a low wattage and it obviously likes more, but in the past it handled it very well at 18W.  However, I did try setting it to 24W to see how that would do and curiously I got almost the same result.  It would display a lower wattage than the 24W limit (22.something I think it was) and then heat up just as slowly and still struggle to maintain temperature.  In fact, I even tried setting it to 30W just to see what would happen and it displayed 28.something and heated up only a little bit faster than the previous attempts.

I wonder if it's even actually correctly detecting wattage or if it's actually far lower than it thinks it is.

Ralim, 2020-01-18:
@Nazosan 

Your edit makes sense, I should probably wrap some more guards around the settings struct 😴 

All power figures in the firmware are complete estimates :) The hardware has no ability to actually measure the power consumed. The slightly under readings are probably from multiple functions rounding down being multiplied.

I'll have a look and see :)

Nazosan, 2020-01-18:
Yeah, I left the post in case there might be something of use in there.  And I think it wasn't necessarily my accidental cross-flash that was at fault either.  I saw the 2.08.1 release out today and decided to go ahead and try it.  This time I did not accidentally cross-flash but properly put on the correct TS80 firmware and when I pressed the button to enter soldering mode it stayed at 0 watts and would not heat even though the voltage properly went up to 12V.  When I did a factory reset it worked correctly.

closed 2020-05-29 by Ralim