Ralim/IronOS #465: DFU 3.45 FW. 2.18 - File too large for the destination file system

guitarmate, 2019-04-24:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**
when copying file to TS100  i get message --- The file 'TS100_EN.hex' is too large for the destination file system. My ts100 shows up as BEDDC4F0(F:)
same  statement if I send a bmp file
* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**
Windows 10
On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

whitehoose, 2019-04-24:
Casting my crystal ball ... my prediction is that the first question from "they who know stuff" will be ...
"What are the sizes of the two files you're trying to transfer"?

guitarmate, 2019-04-24:
I was the firmware from the github site. Which I can no longer access.

guitarmate, 2019-04-24:
Also can't load 86k file from uy chan.
The windows directory shows plenty of room like over a Meg.

closed 2019-04-24 by guitarmate

ppisljar, 2019-06-29:
did you find the solution for this ?
i have similar issues, when connecting drive shows up as BEC8AD26, it has `._.Trashes` file on it.
when i try to copy ANY file to it (including HEX) it reboots immediatly, windows says `The file 'TS100_EN.hex' is too large for the destination file system.` or `Destination does not exist`  ... seems to be related to the device rebooting before file copy can even start.

my dfu version is 3.45 

i can't check the firmware version anymore, since after this i ts100 no longer turns on. plugging 12v in doesnt do anything until i press the button closer to the tip, where it again goes into bootloader it seams (dfu 3.45 shows on screen)

whitehoose, 2019-06-29:
If you are seeing a wastebin you're not in DFU - the iron is displaying as a USB drive.
The 8 digit ID is fine - you might see "sys volume info" ... but no wastebin.
The sequence is hold button 1, plug iron in to power (while holding the button in) . Iron starts in dfu mode o/S opens new file window ... no wastebin
Drop firmware into window - WAIT ...  The iron will cycle and show .err, then auto restart.
File window closes (WAIT for it!) thinks, window opens.
Drop F/w file into window again.
WAIT
File window closes / opens (eventually - be patient) / update should be completed.
3-4 minutes - feels longer