Ralim/IronOS #467: Changes to temperature calculations

Kubuxu, 2019-05-01:
Hi @Ralim, I have changes in works that I would first like to run through you and hear your opinion before spending time cleaning them up.

The biggest would be to switch from raw sample to a temperature in Celsius in fixed-point format for calculations of the temperature in the temperature controller.

The other would be to move away from linear approximation to thermocouple models in Look-up Tables. I have generated LUTs for many thermocouple types (TS100 uses K type, I think Hakko uses N type, I have modes also ready for C, G and D types). I've used Matlab to perform piece-wise linear fit and thermocouple polynomials from https://pypi.python.org/pypi/thermocouples_reference).

As I mentioned in another issue, the temperature reading is very accurate if core and tip temperatures are allowed to equalise by isolating it from the outside (I've used glass fibre and cardboard to do that, cardboard by itself got a bit smelly at 300C).
The temperature delta in the open air is about 20C but it can be compensated for.

You can see a very rough version of the code [here](https://github.com/Ralim/ts100/compare/master...Kubuxu:feat/lut?expand=1). It is still very much WIP. The temperature controller is still a bit buggy.

The flash size difference of using fixed point math and LUT vs master is actually negative (by 20 bytes) and each thermocouple model needs only 32 bytes (I think other types than K and N are rare in soldering irons).

I also plan to switch this code from handwritten fixed point to C++ header lib but this shouldn't add any flash overhead.

---

Note for the future, to myself and others: don't do 64bit division in Cortex-M3, the division routine is huge. This also means that one can't do divisions with two fixed points that fit in int32 (where both are runtime) but it can be easily worked around with minimal precision loss.

Ralim, 2019-05-01:
Hi,
I would say you could probably open the pull request whenever you wanted, and just tag me when you think it's ready to merge.

I think this is a good direction, and the code is honestly fine. I would prefer the fixed point either moved to its own files (purely to keep it cleanly seperate) or to use an existing library. 

The temperature controller shouldn't require any large changes to adjust for your changes I think. What kind of instability are you seeing in the temperature controller ?

I *think* you could change your struct to a constexpr to make it become only loaded into flash, not sure if your current one is automatically being packed in or not, I'll grab the map files and check when I have time.

Keep in mind with the temp controller that, that task is running on a fixed schedule from the ADC controller, the faster that function runs the faster the UI etc updates and makes everything feel fast. So fixed point is 100% a good idea, and I don't quickly spot any big issues in what your doing :)

Kubuxu, 2019-05-01:
>  What kind of instability are you seeing in the temperature controller?

No instability but when I've readded the `miliwatsHistory` part back (I component) it started driving the iron too hot by about 20C while reading low.

```
arm-none-eabi-nm --print-size --size-sort --radix=d Hexfile/TS100_EN.elf | grep Lookup

Ralim, 2019-05-02:
You may need to adjust the tip thermal properties (such as mass) and also how far back the history is keeping averaging for.

That looks good that its being compiled down properly :)

Ralim, 2019-12-29:
Closing this as @Kubuxu's great start has lead to the latest temp re-writes :)

closed 2019-12-29 by Ralim