Ralim/IronOS #469: stable releases?

Stormwing01, 2019-05-12:
I'm just curious, as I'm getting my own ts100 soon... what release is the most stable so far? I dont really have a problem with using older releases, but I don't want to brick or damage my iron some way or another like some of the issues I have read here in the issues section.

Thank you for any advice!

Ralim, 2019-05-13:
Its generally really hard to break an iron unless its defective from the start.
The latest RC is an RC because its not a perfectly happy release, but I havent had time to tune the system a bit better to be happy with it + there is an incoming update to the thermocouple modelling that im inclined to wait for before looking at tuning more.

So, i would try the latest RC but if you dont like it, go back to the latest stable.
Nothing should break, and if it does, just post here and we will try and help you :)
Generally, your actually quite safe. With the roughly 1-2k hits to the releases page per week, or 6k views per week; that this firmware gets, there have only actually been less than 10 "damaged" units turn up, quite a few are people not reading directions when flashing the firmware, not damaged units (it doesn't break, it just refuses to load the firmware).

Generally if the hardware fails prematurely, almost all vendors are great at helping with that too :)

whitehoose, 2019-05-13:
Generally they are logically stacked, the latest fixes previous issues or adds new facilities.  In general terms the latest is always the greatest, although with windows we'd routinely work with a month's lag to pick up serious problems, these days that's down to a week in most cases (not server).  With one exception (a reported tip overheating - which I don't remember anyone else experiencing) instabilities are minor, immediately obvious and easily reversible by just installing the previous release.
If it's a real concern (not for me) give it a week before you download and read the posts before you jump.

Worst that seems to be out at the moment is a discrepancy of a few degrees between displayed and actual tip temp - hardly a real world problem to anyone who actually works with irons (temp displays are still the exception rather than the rule) - you work with what you observe not the display.

Stormwing01, 2019-05-13:
Thanks you two, you put my mind at ease and I'll definitely let you know if I find anything out of the ordinary. It might still be a week or two before my iron actually arrives :/

closed 2019-05-21 by Ralim