Ralim/IronOS #471: Missing elements on lcd while heating [TS80]

DjordjeMandic, 2019-05-17:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
When i power up the iron and start heating up the tip the "heat going away" animation is missing. All that i can see is just a rectangle that should represent the tip.

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Power up the iron.
2. Start heating up
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06
  - PCB Version: (1/2) (not sure, it's displaying model 2)
  - Power Supply (Voltage and Current Rating) : 9V 2A

DjordjeMandic, 2019-05-18:
It would be nice for someone who has ts80 to check this also.
[image]
I think that there's something wrong with my iron not the firmware.

whitehoose, 2019-05-18:
I'm home watching Taken 3 and my power and kit is upstairs so I've only worked with 5.2v usb2 on my old laptop.
Started by re-flashing using linux mint.
Got the usual double dip.  I don't remember after the second copy file manager showing both the .err and the .rdy files (120k) but on the restart both files were deleted as expected and iron working OK.  

Threw me a little, I've never used the simple display in anger. I thought if I'm using a different, better firmware I want all the bells and whistles available and on display.  Plus I'm pretty sure that the overheating glitch only struck standard display, but yes I get exactly the same display you're showing,
I'd have guessed battery indicator - but I was on usb!

I've reverted all settings to simple - reset - then heated the tip to melt solder, slow - but all worked as it should.  Back to advanced mode and all looks fine.
I'm using a new downloaded version of  v2.06-RC4 just in case

DjordjeMandic, 2019-05-19:
@whitehoose there should be heating animation but its not working, i looked in source code and something smells fishy. it must be firmware bug.

whitehoose, 2019-05-19:
I wonder if the anim's been ditched?  The moving numbers tell you what's going on.  In detail mode (which I use normally) you get so much more inf - which is nice.  I have to admit I don't believe any of it though.  It's good to know that a higher number means a hotter tip - but whatever the numbers are - if the job doesn't look and smell right they are only reference indicators.  Who ever looked at a job and thought that's a 324.76c job if I ever saw one.
My head says, "OK the thingy is a bigger chunk of metal" - So I know I need to pour on more pixies to make the glue work.  Experience gives me an idea how many - but practice makes it exact (eventually).
It's why I'll never be able to take an iron with memory, a display and Processor seriously!!!
S' like a rubber hammer with internet.

DjordjeMandic, 2019-05-19:
Its not ditched, its clearly in the code but it does not work, i wonder why.

whitehoose, 2019-05-19:
Dunno ... we've had display misalignment, language glitches sometimes with the best will in the world - sh*t happens. ralim and the others are usually fast to respond ... maybe he/they've all gone walkabout or have something important on.  I'm sure in the fullness of time .... in the meantime, have you tried detail view?

DjordjeMandic, 2019-05-19:
Detailed view works fine but now im confused a bit because ts80 on original adapter(9v 2a) pulls only 16.2W while on samsung powerbank eb-p1100c(9v 1.67a) pulls 17.5W.

Ralim, 2019-05-21:
The power reading is an estimate based on tip resistance, measured voltage and estimated PWM duty cycle. Its not "accurate" but instead intended to give a ballpark.

I'm currently snowed under in life outside of github, so im reading issues but struggling to get time to work on things. Most likely this broke in the change from 0...100 to 0...255 PWM range, Should be fairly easy fix if someone wants to read the code and make a PR, but if not ill get to this when i get time (soon i hope).

DjordjeMandic, 2019-05-21:
@Ralim Take your time, i was just scared that my unit is faulty. Thanks!

Ralim, 2019-12-29:
This should be fixed in the latest release (I hope), please re-open if its not fixed and ill check again.

closed 2019-12-29 by Ralim