Ralim/IronOS #473: Cant flash firmware

feivur, 2019-06-06:
- [v] Bug report
  - [ ] Feature request
  - [ ] Translation

I press button nearest to the tip. Plug usb cable. On display appears DFU:3.43. On computer appears disk with name C4DFCDEF, 2Mb size, 2Mb free. Then I drag-n-drop v2.06-RC4 .hex file to this disk. Then disk disconnect from system and appears message about file too large to this file system.
After this iron not boot up when i connect the power. Same behaviour on any firmware version.

Ralim, 2019-06-06:
Hi,
Could you please confirm which operating system you are using?
Have you searched the previous GitHub issues raised about this? Almost always this is due to how your computer is configured / setup

feivur, 2019-06-06:
I checked os version:
Windows 10 64bit, build 1809

feivur, 2019-06-06:
I tryed on other PC with Windows 10 64bit, build 1803, and there is no problem, iron successfully flashed. 
Dont know, what was cause of fail on first PC. Close issue.

closed 2019-06-06 by feivur

1iggy2, 2019-08-19:
Same issue Windows 10, Version 1903. The file explorer from the TS100 closes immediately. Attached is image of error message, file name, and Windows Update Page.
[image]

1iggy2, 2019-08-19:
Tried on another identical system. Same windows version, same cable. Worked like a charm. I don't know if this will help troubleshooting this issue in the future if needed, but the computer that worked is an Asus Laptop, been in use for 2-3 years. The computer that didn't work is a desktop built in May 2019, B450 Tomahawk, Ryzen 2600X, RTX 2060, 32 GB RAM.