Ralim/IronOS #476: feature request: display invert and slower change

bl4k1st, 2019-06-20:
* **I'm submitting a ...**
  - [ ] Bug report
  - [X] Feature request
  - [ ] Translation

Dear,
I have two feature requests:

1) 
detail display of standby is too similar to detail display in soldering mode.
Would it be possible to add an invert option, so standby mode is mostly white with black letters?

2)
the temperature number in standby mode (normal, not detail) is changing very fast. Would be a settable slower change possible, e.g. every 0.5 seconds, or 0.75 or 1 second?

Thank you,
awesome firmware.

jj-uk, 2019-06-25:
#1 is possibly related to https://github.com/Ralim/ts100/issues/464.

#2 I'd consider as a bug.

Ralim, 2019-08-27:
@bl4k1st Would you be able to have a quick poke at my last message in #416 and see what you think of that ?

v01d, 2021-02-03:
I think that inverted screen would serve as a nice quick indicator to let you know that it is on and hot. I'm used to look at the red blinking led on my soldering station and reading the oled is not exactly the same

discip, 2022-04-23:
@Ralim
Since the OP does not seem to be interested anymore, this can be closed. 😊

thanks

closed 2022-04-23 by Ralim