Ralim/IronOS #477: oLED over drive, early panel death?

charun-za, 2019-06-20:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
"Possible Bug"

* **What is the current behavior?**
OLED lost 90% of all brightness and wont display at all with your firmware, functional however(heating...etc)
Original firmware "works" fine but brightness is still down but you can read the display, still functional.

* **What is the expected behavior?**
So one I had one of my ts100 irons display pretty much die (very very dim atm and wont display anything with your firmware) but will still heat and change temperature, when using original firmware it will display fine but the damage is already done (so its very dim) 
I think but could the increased update/refresh rate? perceived over the original firmware, it seems a lot faster on your firmware which is awesome however I think the oled panel used is being over driven and killing it? correct me if I'm wrong however. 
If possible could I have a modded version with original refresh rate dropped a bit? maybe 20hz or 30hz or lower depending what yours is currently set to, if you have access to change that all..

let me know if you need more info.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. use the firmware :P?
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2) 
  - Power Supply (Voltage and Current Rating) : 19~20v

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-06-20:
Hi,
Firstly sorry to hear about your iron :cry:
All OLEDs have a finite life, and these being cheap low cost units you may have hit the end of life your your OLED.

This firmware doesn't use different screen settings to the stock firmware anymore, except that it checks for errors when transferring to the screen and aborts properly.

The OLED is communicated with over a i2c bus in bursts of screen updates. , This is no different to the stock firmware really.

The OLED then updates it's multiplexed screen internally at a high refresh rate ( I think ~30hz).

Sending screen updates faster than that to the screen doesn't result in any faster updates to the screen.

In my firmware an entire screen updates event occurs every 10 milliseconds or so, so nothing crazy fast.

I just don't down filter the displayed temperature as much, so mine looks faster as it shows you the actual readings. There isn't much of a functional difference in the way the screen control works, certainly not enough to cause overdrive. (The settings I use are from their firmware, just with a second varient that inverts the scanning direction to help with screen rotation.

I could change the screen refresh rate but generally that doesn't affect much in terms of screen life as it's still being internally refreshed.

charun-za, 2019-06-20:
Hi Ralim, thanks for replying :)!

Yeah it is unfortunate. Thanks for the insight on this. So 10ms is around 100hz then? but for something this small its not that fast. but that doesn't affect the multiplexer, as I guess it runs on its own frequency. 

Miniwarwe did send me a replacement iron, so no problems, I understand oled have a life time but its in the scope of years.... not like maybe total on time of a few days like mine was, otherwise they would've even attempted to market the tech ,but that's neither here nor there, the panel was probably defective to begin with although nothing obvious when I first used it and using your firmware was just coincidental. 
interesting you say it uses i2c bus, I did get a 128x32 0.96" panel replacement for a local supplier but the pin out is completely different, still i2c bus, and the resolution is lower on the original, (96x16)
I think i would be able to figure out the pin out of original;
( https://www.buydisplay.com/download/manual/ER-OLED0.69-1_Series_Datasheet.pdf )
as I think its this one....anyway the hard part for me anyway would be get this to work in software side for the image scalier or maybe the ssd1306 display driver might do it for me? any thoughts on this~

Ralim, 2019-06-20:
Yeah, it's not overly fast all this concidered.
One of the Chinese data sheets I've found for this screen is like 5k hours or so.

If it's faded out, it might be possible that it wasn't sealed properly when it was made.

Yeah the screen is an odd one, I've found them on AliExpress before, but they are not used on breakout boards so not as common. The pinouts is in the schematic and I would suggest aiming for a compatible one if you can, or else you will need to maintain your own fork :/

The driver won't scale it, most likely you will end up with the image squashed onto one half of the screen. You could probably mod the software to double the size of the buffer and pixel double every draw event, but it would be a non trivial change as I draw everything by directly memcopy'ing the buffers.

charun-za, 2019-06-20:
Cool Thanks!, 

I will be experimenting see what can be done by hack something together~ fun times ahead xD

closed 2019-06-20 by charun-za

jypma, 2019-12-28:
@charun-za Did you ever end up trying the 96x16 screen from [buydisplay](https://www.buydisplay.com/graphic-96x16-i2c-serial-blue-0-69-inch-oled-display-mp3-module-arduino)? I'm having a dim OLED screen (burned-in?) and am looking for a replacement that would "just" work with this firmware.

charun-za, 2019-12-30:
Hi @jypma , I did not, ;< but looks like the pinout is the same, so i guess the linked display would work, you'll just need to try it, otherwise if your iron is still within warranty you can email mini-ware claim it through them; http://www.miniware.com.cn/contact-us/ on miniwares website, they sent me a new one, let me keep the old one so i might do it one day, when i order the display.

jypma, 2019-12-30:
Thanks a lot for your reply!

I reached out to mini-ware, haven't gotten a reply yet. But since the display is so cheap, I ordered one (off [ebay](https://www.ebay.com/itm/White-Graphic-Serial-SPI-I2C-0-69-96x16-OLED-Display-Module-for-MP3-w-Tutorial/302039712958) because lower shipping). I'll let you know what pans out :-)

jypma, 2020-01-17:
Well, good and bad news: the [ebay] (https://www.ebay.com/itm/White-Graphic-Serial-SPI-I2C-0-69-96x16-OLED-Display-Module-for-MP3-w-Tutorial/302039712958?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) model _does_ work, but it seems its flat-flex is the wrong side up: I had to solder it on the _underside_ for it to fit. Luckily there were some vias near the connector side, so I could solder it in place. 

I wouldn't recommend that model to others considering it.

Miniware didn't respond to my inquiries.

charun-za, 2020-01-19:
nice man, yeah i suspect it might be hard to solder considering its size and the type of connector it is, but does it work ?

daf33, 2020-01-30:
Hi  I'd be interested if it worked too as mine has just gone dim aswell

jypma, 2020-01-30:
Yes it does work :-) My process was more or less the following:
Needed tools: Hot air gun, soldering iron, kapton tape, pincet, copper braid

- Mask off non-connector areas with kapton tape
- Evenly heat up the connector area with the hot air gun, until it's _really_ easy to push it off the PCB with a pincet. I was too quick here, and pulled several pads off the old connector (and the PCB). Be gentle here.
- Remember precisely how the old screen fits
- Re-tin the PCB, use the copper braid to clean off the tin and have clear pads
- Get the new screen and check how it should be folded and fit at least 5 times, so it's not upside down :)
- Tin the right side of the connector. If you use the bottom side (I had to), "scrape" the vias a little so they take tin
- Place the connector on the PCB, and heat it one area at a time with the soldering iron, applying downward pressure with your pincet to keep the connector down as the solder hardens.
- Test and smile :) (or not...)

charun-za, 2021-10-05:
so i just found some oled panel/screens got 2 of them and managed to solder a new one into, works 100%, initially had to solder it twice, bad contact i suspect and my terrible flux which needs to be cleaned otherwise it causes high resistance shorts just though i'd just update my progress and successful repair, spare ts100 now :D
https://puu.sh/IggX6/23101c63bf.jpg