Ralim/IronOS #479: heat up time

ppisljar, 2019-06-29:
I am wondering what is the average heatup time ?

i am testing TS-1 tip, running of 12V 250W power supply and it takes around 30 seconds to reach 300 degrees C from the room temperature. Is this normal ? (i was under impression it should be under 10s)

Ralim, 2019-06-29:
Hi,
12V is the bottom end of the usage voltage range. 
Power in the tip is P=I*V, I=V/R, Therefore P=V^2/R. R (resistance) is approximately constant.
So the power going to the top of the iron is proportional to the supply voltage squared.

At 12V your maxing out at around 17 Watts into the tip, which is why it's so slow.

At 24V,  the tip is running at a maximum of 70 Watts instead. Which is where the figure of heating up in a couple of seconds comes from.

ppisljar, 2019-06-30:
thanks a lot ! i tested with 24V supply and the same tip heated up to 300C in around 4 seconds (overshoots) and needed around 4 seconds more to completely stabilize on 300C

closed 2019-06-30 by ppisljar