Ralim/IronOS #481: Bricked after FW flash attempt, other fixes have not worked

Lachkopf, 2019-07-11:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [*] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a possible bug

* **What is the current behavior?**
Brand new TS100, tried to copy latest release .hex to it. "Unexpected error" popped up as the TS100 disconnected. After i tried moving it and repeating with different .hex versions (unsuccessfully), i tried to power it up to do some work. When connecting to Power, TS100 no longer boots or shows anything. No further attempts solved the issue.

* **What is the expected behavior?**
Copy .hex to TS100, get .RDY file signaling successful flash of new fw. Powering up normally with included power brick

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Connect TS100 while pressing button closest to tip. Windows shows
***Steps to reproduce:***

1. Connect TS100 while pressing button closest to tip
2. Copy/Move .hex fw onto the TS100
3. Fail to flash .hex

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
Not really sure what this question means.

* **What are you running:**
It shows DFU 3.45
On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.05 or 2.06 were the ones i tried
  - PCB Version: (1/2) How do i find out?
  - Power Supply (Voltage and Current Rating) : Output 19V/2,1A

* **Other information** 

If submitting graphics to go on the iron, please use BMP oConnect TS100 while pressing button closest to tip
[image]
r PNG files over JPG.

FFY00, 2019-07-11:
Flash the official firmware.

Lachkopf, 2019-07-11:
> 
> 
> Flash the official firmware.

As in the stock bootloader? Tried that, but it didnt fix or change anything

Lachkopf, 2019-07-11:
Solved! Tried flashing the file with my Laptop, worked like a charm.

closed 2019-07-11 by Lachkopf

whitehoose, 2019-07-11:
Can't leave a comment (officially) but!  What folder are you saving the fw in?  Windows 10 is very keen on executable files that aren't in the w10 white list running within system folders.  Some virus checkers not only support this but extend the definition of perilous.

I usually create either a personal folder in the root of C:, A second partition on a single disk or best, install a second BIG storage device and leave windows and it's new 3 partition structure to its own company.  That usually does the job (you should be able to prove the concept with a USB stick). If needs be open file manager with administrator rights (and of course your profile should be administrator not a standard user too).
Finally - a really keen virus package may need disabling for 10 minutes.
My w10 loads f/w without problems, as does linux and win 7 (I have 3 way boot).
a ssd partitioned for OSs and a hybrid as storage.

yaamin6236, 2020-04-15:
> Solved! Tried flashing the file with my Laptop, worked like a charm.

Thank you so much, at this point I thought my ts100 was broken as it was only working in DFU mode and when trying to upload the hex file on my desktop it wouldn't work. After trying it on my laptop it worked perfectly :)