Ralim/IronOS #483: Minor graphical issues in RC5 (TS100)

Fluvance, 2019-07-16:
* **I'm submitting a ...**
  - [x ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
Incorrect power reading.
Using a TS100. In RC5, I am seeing the iron report a max power usage of 3.3W. It only shows 0 or 3.3, even during initial heating. Typically, in RC4, it would be showing around 58W at this point. I have also verified with an external power meter on the supply. The iron takes approximately the same time to heat up as RC4, 8 seconds.

Display measurement update speed.
Also, in RC5, I am seeing a much slower update speed on the temperature and power displays, maybe at around 3-4 updates per second. In RC4, the screen updated much faster, around 8 updates per second? I'm not sure if this was changed due to some past requests, or if this was inadvertent. If it was on purpose, would it be possible to have an option to change the update speed? Purely for aesthetics, but the iron felt more premium and more powerful when the numbers updated real-time.

* **What are you running:**
TS100
2.06-RC5
PCB Version 2
Power Supply: 24V 3A

* **Other information:**
Just wanted to say thanks for the crazy awesome work you put into this!

Ralim, 2019-07-16:
Hey 🎉,

So there were some timing tweaks for usability to stop people complaining it updates too fast.

The internal PID went from 32 to 16Hz, which was mostly for stability reasons on hardware where the tip is higher capacitance than normal. (Allows me to extend ADC capture time  but maintain same max power).

The temperature displayed on the screen also gained a rolling average filter which is what your seeing. The screen itself still updates just as fast, but the firmware is trying to smooth out the readings a little bit :) 
Because I'm using templates to generate optimised for for the filter it's a bit of a pain to make it a user selectable option  but it might be sped up a little bit.

Power reading is a bug, and I'll look into this when I get time, I did refactor a bit of the PID stuff, mostly trying to tune it a bit better for the TS80. Apparently I broke that estimator so I will try and get a fix soon :)

Fluvance, 2019-07-17:
Ah yeah and now of course I'm complaining about it being too slow! Can't please everybody. I didn't notice it before but the averaging really helps with with the temperature accuracy.
Thanks for the response!

Ralim, 2019-08-27:
Happy to help!
I'm going to close this issue and track power meter over in #485

closed 2019-08-27 by Ralim