Ralim/IronOS #484: Configurable wattage on TS80 with nonQC power supply

kubena, 2019-07-18:
* **I'm submitting a ...**
  - [ ] Bug report
  - [x] Feature request
  - [ ] Translation

* **Request:**
Would it be possible to configure current limit (amperage) on TS80 for legacy USB (=non QuickCharge) power supply? 

Currently it is set to 1 Amp (5 Watts). As a last resort it is OK, however my measurements suggest, that even very slight increase in maximum current would yield significantly better results. Since many chargers and powerbanks support 2Amp+ charging over legacy USB, it would greatly improve usefulness with non-quickcharge power supply. 

I suggest configuration setting in advanced menu which would allow to set 0.8 to 2 amps with 0.2 step would be great (4W to 10W with 1W step). 
If too complex, even configuration option in codebase would be greatly appreciated.

With 5 Watts it is only able to reach 300C at maximum and it takes around 5-10 minutes (on QC it uses around 5W @ 300C while idling). With only minor increase to 6W it would reach around 340C (on QC it uses 6W @ 340C) and it would reach usable temperatures at around 300C much quicker.

whitehoose, 2019-07-18:
Ralim may well correct me here but as far as I'm aware amps and watts don't work like that.
The Stock firmware just errors out with Low Volts.  That Ralim's FW entertains USB2 at all is quite impressive (and very handy in a panic). 

Even cheapo PSUs output a (usually) surprisingly well regulated voltage and have a well defined current limit.  Plug a dummy load and tester into a usb port and wind up the load - You can see the port shut down when you exceed it's current limit (the kit is cheap and handy to have if you use powerbricks).

The tip has an expected resistance (load) and the amps and watts occur as a result of those factors (V=IR or w=IR).  It's not negotiable.  You can change circuit resistance by using a different lead tip or powersource (this last one is going to have minimal effect on its own).  Unless you are using appallingly poor (or ridiculously long) components the connecting cable resistance should be >1ohm.

You can make power available - but that's no guarantee that the circuit will take that amount, Ohm's law trumps wishes.  Unless you can alter the characteristics of the circuit tings won't alter.  

Different USB sources have different limitations - but unless you've put something together and connected it to the mains the USB standard states that usb2 can deliver between 500mA to 2 amps (I think) HOWEVER - the manufacturer of the PSU will either be relying on crapppy batteries or sophisticated circuitry to regulate the output voltage to (again from memory) 4.5-5.2v and will state the intended current limit per outlet (my cheapest is 1x500mA).     

My ts80 will melt solder using a USB2 source - but it takes <3 minutes to get there, only my PSU will deliver the pixies when set to usb2  - my single cell power brick (more of a pebble) can't deliver anywhere near enough current.

Once I switch over into USB3 mode ... the Iron then takes off ... but it needs QC3 capabilities (and that extra 4v) to do it.

kubena, 2019-07-18:
Not completely sure how it works, but I do have several USB power supplies capable of 2A@5V and iron never takes advantage of it (always draws 5W) while it can take advantage of 2A@9V(around 18W) and 2A@12V (cca 24W).
Since I cannot see how those three very different values could be directly tied to single internal tip characteristic, I assumed it is negotiated value and not directly related to tip resistance.

If it is hardware limitation, then obviously it will not be possible. Anyway thanks for the answer.

whitehoose, 2019-07-18:
If you assume 5, 9 and 12v (not certain 12v is an option on the ts80)
and a 4 ohm tip
current=(v/r) so 5v/4=1.25a, 9v/4=2.25a, 12v/4=3a
but power = v squared\r
so 5v=25/4=6.25w, 9v=81/4=20.25w and 12v=144/4 = 36w !!!!!!
Little increases give BIG changes - double the voltage gives almost 5 times the power
if the iron is rated 18w - at 9v that's 2A, tip res =4.5 ohms.
at 5 v to make 18 watts needs 3.6A with a tip res of 1.4 ohms.

it's going to melt down

Ralim, 2019-08-27:
Sorry for noise, mis-read the title.

@whitehoose is correct here, as we have a 4.5 ohm tip, we cannot exceed ~ 1A at 5V. As the power drawn is proportional to the square of the voltage.

@whitehoose Any chance you feel like wrapping up your in depth explanation into a PR for the power.md file ?

closed 2019-12-29 by Ralim

whitehoose, 2019-12-29:
> @whitehoose Any chance you feel like wrapping up your in depth explanation into a PR for the power.md file ?

Ralim
I've only just seen this comment.
I'll give it a go if you still want it.  I'm concision that my "knowledge" is basic general pixie wrangling and not product specific - also I do have a couple of ts80s - but not a ts100.

Ralim, 2019-12-29:
@whitehoose 

Even if you could just make a draft PR, would be a great help. You have chimed in, on quite a few conversations and I would like to start pulling these into files in the repo rather than scattered throughout github issues if possible :) I'm happy to do cleanup / organisation, just haven't had time to start recently. _Trying_ to get get on top of it.