Ralim/IronOS #485: Incorrect power meter display v2.06-RC5

NeoVerter, 2019-07-18:
* **I'm submitting a ...**
  - [x] Bug report

The power display fluctuates around one watt when warming up. When the temperature reaches, the indicator shows zero or a very low value.
Powered at 19 Volts.

TheUntouchable, 2019-07-19:
Same problem here with an TS80.

Using a Keweisi KWS-MX18 for power supply monitoring an a charmast W2009P powerbank wich is able to supply 12V. The TS80 is also set to this value.

First the TS80 is only using 9V, this is something I can see on the powerbank, the KWS and the iron itself.

Second when heating up, the iron is telling me it uses 29W, the KWS-MX18 is telling me it only uses about 18W. After the heat of 450° is reached, it will not stop heating and hits over 470°, the iron is telling me its using between 0W-90W, the KWS is telling about 7W. After some seconds it will slowly decrease the temp around the 450°.

CutterSlade001, 2019-07-22:
Same, TS100 never shows over 3.1W even in boost mode. 24V 6A notebook style AC adaptor.

TheUntouchable, 2019-08-02:
Nothing new on that? I will go back to the old firmware es the iron starts smelling after it got over 480°.. :(

Ralim, 2019-08-02:
It's purely an aesthetic issue with me forgetting to change a divisor before the number is shown. It doesn't affect operation.

The irons tip is only rated to 400C for main use, and 450C is pushing it to the limit.
You shouldnt use the tip above 380C for prolonged use.

Ralim, 2019-12-29:
This should have been fixed in the latest release

closed 2019-12-29 by Ralim

CutterSlade001, 2020-01-07:
2.07:  Just checked it out. Can confirm, it's working again, thanks for your hard work! <3