Ralim/IronOS #489: TS100 Temperature Calibration

Bittronics, 2019-07-24:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**

After calibrating the TS100 the iron says the the tip temperature is at 240 degrees Celsius while in reality it's only at 15 - 18 degrees 

* **What is the expected behavior?**

Calibrating the tip to display the correct temperature

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 19.4V 3.5A

Ralim, 2019-08-27:
This has been an ongoing issue for a while now, and I'm honestly thinking of re-working the temperature measurement code entirely than keep poking this calibration issue as it stands.

I.e. go for a much more accurate temp curve and processing and simpler calibration.

Ralim, 2019-10-08:
FYI, I'm re-working the calibration and temp control stuff in a branch.
Will be more news on this *soon*.

disaster999, 2019-12-15:
Hi, not sure if the calibration is still an issue or not. I just got the TS100 and updated to the github firmware. I was messing around with the settings and thought I'd calibrate the temperature. I follow the instructions on the screen and made sure the tip was at "room temp" and pressed the button closest to the tip. 4 dots appeared and it went back to the "calibrate temperature" menu. I pressed that a few times and all did the same so I exited the menu

I enabled the detailed idle screen so I could see the tip temp and found it is at 31, 32c....when my room is only 20c. Did I screw up or didnt to the calibration right? Does this mean the temperature read out is always 11-12c higher for me?

whitehoose, 2019-12-15:
Bit of a disaster that.  Calibration relies on the whole iron being at room temperature - the calibration just happens (nothing to see here folks) ... except that is, being grasped in your hot sticky expectant hands - it starts to warm up ... the more you try the nearer to 37c so all you can really say from your description is you don't appear to have a fever.

Next time you need to be quick and only do it once ... then you leave it for an hour before powering on to see the starting temp (you can try some big thick gloves to keep IT away from YOUr heat) but leave it alone while things normalise again.

Personally - if you are even 20c from actual, its not bad.  What really matters is what's happening on the tip - if the glue doesn't melt - no matter what the number says you aren't hot enough.  The glue will melt (63/37) at about 187c - your working temp needs to be about 300-350c for the magic to happen.
If the flux vapourises and hits you in the eye - you are too hot.  The flux will burn and you'll see it turn black and nasty.  Reduce the numbers a bit and try again.
Too cold - and the magic molecule meld won't happen and the solder won't merge with the copper.  The solder will act like putty - you'll have a dry joint and that will fail at some point (if you know what that looks like you don't need anything else).
... too hot and you could damage the components or the tracks - the joint will be brittle ... and its going to fail.
You want a steady melt that climbs slightly upwards before setting.

disaster999, 2019-12-16:
sounding kind of hostile there...I mean no disrespect. Simply questioning if I calibrated my iron correctly or not.

I wasnt holding onto the iron in my "hot sticky expectant hands" like you suggested and simply had it resting on the soldering stand for an hour like you had suggested and the calibrated temperature is still at 30+c

This kind of defeats the purpose having an actual temperature calibration option when you cant actually calibrate it. And going by your little guide for me on how to use a soldering iron, temperature numbers means nothing as long as whatever "number" you set on the iron, as long as it melts the "glue" your fine.

Ralim, 2019-12-16:
@disaster999 

I don't think @whitehoose intends to be hostile. This is more of an old topic that has been discussed in many different issue tickets. 

Fundamentally it comes down to the following points:

1.
It generally doesn't actually matter what the temperature is for most soldering, just that its stable and repeatable. (And that its somewhat accurate). You can even see this shown as most digital irons "hide" their fluctuations and only show these after a certain %age of deviation.

2. 
We cannot change the hardware in the iron itself. This limits us to just the tip and the cold junction temp sensor.

3. 
We have limited options in the devices in the iron to measure a known reference. Specifically, we _only_ have the temperature sensor in the handle

4. 
The default for the firmware is a _best guess_ for most irons, that should typically land you within a few degrees.

5.
There has been lots of experimentation of different tips, and generally every tip is different (model to model, batch to batch), and every iron is different (unit to unit).
However, these variations are typically a few degrees.

6. 
The tips response is non-linear, and not even nicely piece-wise linear in sections. So there is currently a branch that im slowly poking _when i have time_ to move to using lookup tables calculated for the Hakko tips vs the Miniware Tips. (Slightly different metal's used in construction).

Tl;dr
The default i good enough for most people.
This issue will get closed once the new branch is good enough to be merged (aka it has a calibration curve for Miniware, as it has Hakko only atm).

disaster999, 2019-12-16:
Thank you @Ralim for the detailed explanation. I least I know I the iron calibrated the temperature as intended and due to hardware and software limitation, this is the best we can get so far.

I will continue to use the iron as is. Thanks for the firmware, it definitely has made solding more fun

whitehoose, 2019-12-16:
> sounding kind of hostile there...I mean no disrespect. Simply questioning if I calibrated my iron correctly or not.
>
> I wasnt holding onto the iron in my "hot sticky expectant hands" like you [...]

Sorry if that's how you saw it - I was aiming more towards tongue in cheek certainly not my intention to offend.  Nor do I want to give the appearance of stalking - but If you want my explanation read on - if the apology suffices stop now. 

My point is that the numbers - even if accurate to the degree are almost meaningless as far as soldering is concerned.  "the book" really doesn't allow for temperature variance it's all about maintaining the ideal temperature +/- 10-20c.
Indeed after 50+ years of soldering, I've only had temperature control for 18 months or so.  It was handy on day 1 when I set things up - but I'm not skilled enough to look at a job and decide that it needs a tip temp of 327.05c.... I don't know anyone who is ....  Nor have I ever heard of a set of tables that would give me a number for a tip temperature other than the manufacturer's recommended working temp - or on some very precise specs for scientific instruments.

In the early days of transistors it was usual to take elaborate precautions with heatsinks to protect them from heat damage.  In those days dry joints due to undercooking was common 

I'm reasonably confident I know when I've made a good joint from the look of the melt and the shape of the join, if I'm in a new environment or outside I'll make a practice join or two to "get a feel".  But its usually my call.
I've been using irons of varying complexity on jobs of varying tech requirements since I started work in 72.  Prior to that it was an emerging hobby - and back then an iron was constructed to melt solder.  No adjustments.  If you needed more "oomph" you bought an iron capable of melting solder at about 300-350c but constructed to have greater thermal density, which usually involved bigger physical mass.   You never want the temp to wander too far from the standard spec.  I remember a batch of audio sockets which had a housing that deformed at a temperature 10 degrees below the melting point of solder.  The irony there was the contact springs were massive and needed a bigger than normal iron to apply the solder - we wound lots of wire on the tags connections and soldered the knot. 
In conditions where a heated iron just couldn't ... a blowtorch would be deployed which could easily reduce your work to ash if you were too enthusiastic.

With that in mind, once that's been achieved the numbers are little more than a red or green light to warn of excessive variance.
All you need is to know when you've turned it up or down and whether the set temperature is holding.
If you don't have the experience to see what's happening at the tip the numbers could do more harm than good because they always reflect the thermocouple and not the tip's, the point of contact  or the work area's attained temp.  So without looking at the site you'll never know the state of the joint from the number.   
Excessive heat generally causes more problems than it solves.
If the iron is struggling you need a bigger iron not a higher temperature.

biasedlogic, 2019-12-18:
I second everything @whitehoose said, nonetheless I'd love to have a setting to input temperature offset for the working temperature. While I don't care whether the display is accurate tip temperature or not, I'd like to be able to set my two identical ts100 to the same number to avoid mistakes. Now they are good 20C apart when hot. The thing is, I don't use them for soldering, but for plastic welding and they are great at that, but the plastic is very sensitive to temperature - between soft and liquid and smoking there's not as many degrees 😅

Ralim, 2019-12-21:
@biasedlogic You can always reset settings to have them both be very similar.
 I agree the calibration isnt ideal, but its sort of the best we have without assuming users have accurate thermocouples and are willing to bond them to the tip.

Ralim, 2019-12-29:
@Bittronics  @biasedlogic Please try the latest release too :)

closed 2019-12-29 by Ralim