Ralim/IronOS #490: temperature reading while no tip attached

discip, 2019-08-03:
Good evening,

* **I'm submitting a ...**

  - [x] kind of Bug report (I do not think this is a bug. Its more of a cosmetic thing.)

* **What is the current behavior?**

The device, TS80, also with Ralims latest firmware, shows a temperature reading of about 512°C, if no tip is attached.

* **What is the expected behavior?**

I thought, it would be nice, if there was no temperature monitoring at all, if not tip is attached.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Plug the device in and remove the attached tip, or just plug it in without the tip in the first place.

* **What is the motivation / use case for changing the behavior?**

It might be a touch of perfectionism.

* **What are you running:**

  - Firmware Version: v2.06-RC5
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : 3.6V /   3A
                                                                            6.5V /   2A
                                                                         9-12V / 1.5A

Thank you in advance
kind regards

timecop97, 2019-08-15:
@discip  This would probably require a hardware mod, since there is currently no way to detect whether a tip is inserted. You could take the approach of saying "if the tempurature is over 500, then the tip is removed", but on the TS100, at least on mine, with the tip disconnected, the temp reads as 444C. the UI can set the tempurature up to 450C, so it's possible to increase the tempurature so much that it thinks the tip is disconnected.

discip, 2019-08-15:
Good evening,
thank you for replying.
I unfortunately do not know much about the internal processes of this device, but as far as I understand electronics, the attachment of the tip (in my case the TS-B02) should result in a changed resistance (I have measured 4.4 Ohm.), which should be measurable by the iron.
In short: If no tip attached, the resistance should ideally be 0 Ohm. (I think the real value is greater than 0 Ohm.)
If tip attached, the resistance should be greater than 0 Ohm.

Please correct me, if I am missing something.

Thank you in advance
kind regards

timecop97, 2019-08-15:
@discip if no tip was attached the resistance would be infinite (open circuit). The circuit measures resistance on a scale which ends at around the maximum expected resistance of the iron. as the iron tip heats up, its resistance increases, which is interpreted by the micro as a change in tempurature. 

you can probably prove this to yourself by attaching a small value potentiometer to the iron and adjusting it. you should see the iron tempurature rise and fall(be careful not to turn on the iron feature as that could kill the potentiometer).

 the resistance measurement circuit will only measure up to the equivalent resistance of 440-450 degrees tip, which is why it's kinda infeasible to detect an open circuit. remember, if the resistance measurement could go up to, say, 800c, then it wouldnt be a problem. Since the maximum tempurature is far hotter than any iron could go, we could just say, if the tempurature reads over 600c, then assume the tip is unplugged. 

you could always attempt to modify the resistance measurement circuit to detect higher measurements. This part here 
[image] is what you'd probably need to tinker with.

discip, 2019-08-15:
@timecop97 
Ok, I think, now I have got your point.
The keyword was 'open circuit'.
But I think there is another way to get the iron to perform just as expected:

As you mentioned, the max expected resistance equivalent to 450°C is the limiting parameter, making the detection of an open circuit almost impossible.
But what if that limit would only apply to the input (_By input I mean the temperature set by the user._)?

For example, if I would try to go past 450°C, the Iron would simply ignore that, the way it is doing already. But the resistance measurement circuit would not be affected by the input limit, so that the background processing, would be able to assume that the tip is unplugged, if the resistance / temperature goes beyond that input limit.

I hope I made my self clear enough.

kind regards

timecop97, 2019-08-15:
@discip hey, i understand your point, but what i'm getting at is that the component that converts resistance to tempurature cannot tell the difference between 450C or 500C or any tempurature above that. It's been designed to respond to tempuratures up to 450C and if the tempurature is any higher than that it'll still read as 450C.

 In the case of an open circuit this would appear as a very high resistance in the circuit, but since it can't express that as over 450C, it will read as 450C. If you took the tip and externally heated it to 500C, the iron would still read 450C. This makes it impossible to detect in software the difference between a very hot iron and a removed tip. As i said before, you could modify the amplifier circuit to accept higher tempuratures, or you could add a switch to the circuit to detect if a tip is inserted. In future revisions of the board a current amplifier would also be cool. 

Regardless, you wont be able to do this purely in software, some hardware change will have to happen to make this work. The alternative is to say that on boot, if the tip temp is over a limit, say 425C, then assume the tip is removed.

discip, 2019-08-16:
@timecop97 
Thank you for being so kind, to guide me through.
This is no dealbreaker, but would be nice to have.
kind regards

closed 2019-08-16 by discip