Ralim/IronOS #494: Sleep mode is not working properly

AndroidGX, 2019-08-24:
Hi there,

Just recently acquired a TS100 and wanted to flash this custom firmware but not sure why, the sleep mode function is not working at all on my unit, while it's working perfectly with stock v218 firmware.

I tried the following without any success :

flash stock v218 firmware, reboot and flash v2.05.01 (FR)
flash stock v218 firmware, reboot and flash v2.05.01 (FR), reset settings
flash custom v2.05.01 (ENG) version
flash custom v2.05.01 (ENG) version, reset settings
flash custom v2.06-RC5 (FR) version
flash custom v2.06-RC5 (FR) version, reset settings
flash custom v2.06-RC5 (ENG) version
flash custom v2.06-RC5 (ENG) version, reset settings
Trying different values in sleep mode settings
Trying pure stock min values (originally wanted by the brand) in sleep mode settings

Lost faith trying anything else out :(
Anyone have an idea?
Any help will be greatly appreciated,

Best,

AndroidGX, 2019-08-24:
Looks like I didn't had chance trying both those firmwares as sleep mode is broken on them.
Tried **2.06-RC4** (that someone recommended) and it works well again.
Hopefully sleep mode will be fixed in the next build as for me not having this function is a huge dealbreaker.

whitehoose, 2019-08-25:
If it's a deal breaker, with the greatest respect, you're playing wrong game.
The whole point of working this side of the stock firmware is that the only thing that's certain is uncertainty (OK that's a bit dramatic for a Sunday night - but that's what the hacking, rooting ... call it what you want is all about).  You can't push the envelope without some mishaps. 

And even more exciting is there aren't any guarantees- if you can't sit down with google and work out what's next - you shouldn't be playing.  I've bricked devices, I've unbricked some too ... but like that unmarked box of chocolates someone went on about - you never know just what you're getting.

Ralim and the other dev guys don't get a salary, nor do they get their kicks screwing up ... their payback will be completing the job.  Remember .. it's supposed to be fun all round.  A glitch is just that... a whoopsie.  They need to know - but use a sad face not a mad face when you do.

If this isn't just a blip and the stock isn't what you want either - you bought the wrong iron.

eventually we'll be seeing it all working, I can then move on to something else that takes my fancy, because honestly - as a soldering iron it's an over complicated bag of worms.
The soldering iron that payed my mortgage didn't have firmware - it just melted metal.  Day in, day out for 28 years.  No temperature, sleep, or other problems it just worked faultlessly.
Actually it still does, but not every day now

AndroidGX, 2019-08-25:
Cool story bro!
_(I close to avoid other useless eventual messages, thank you)_

closed 2019-08-25 by AndroidGX

Ralim, 2019-08-27:
@AndroidGX 
Hey man, sorry to hear that sleep mode is not working on the later releases for you.
As you didnt include the default template for the issue, are you able to confirm the PCB model for your unit so I can try and replace the issue?

In the debug menu were you also seeing the issue where the time of movement increments continuously or is it fixed at 0 ?

There are multiple reasons why the sleep time may not be working, from an Accelerometer I2C issue through to it not being detected, or possible your daughter board inside the iron has a bad connection.

AndroidGX, 2019-08-27:
Hello Ralim,

Thank you so much trying to help out, I really appreciate your fantastic work you made for our beloved TS100.

I will be happy to help you my best, can you please tell me how to access this debug menu? 
Once connected to USB, I tried a long time clicking button (from AC side) and I can see « 2.06 TS100 Jan 30 2019 », and « Model : 1 », not sure if I accessed correctly the debug mode. Are you referring to the « Move » option ? I just checked and when I try to move my TS100, the numbers are correctly changing. Looks like I’m not alone to encounter this issue as everybody have sleep mode issues with the two firmwares quoted above. Only the RC4 (from the most recents) looks working again for that sleep mode.

Do I need to flash a specific firmware to get this debug mode ?

Many thanks in advance,
Best,

AndroidGX, 2019-08-28:
Ok I just flashed 2.04.1 and checked, I'm on a "PCB1".
Cheers'