Ralim/IronOS #495: Screen is very dark and shows weird number

SIERRA-880, 2019-08-27:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
I'm reporting a bug

* **What is the current behavior?**
The screen look very dark

* **What is the expected behavior?**
The screen was brighter 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***
1.  Don't use the TS80 for 2 months
2.  Plug it to your power bank

***Video of problem if hard to reproduce***
https://photos.app.goo.gl/AwEPhgnCvKNGfgv66

* **What are you running:**
  - Firmware Version: 2.06
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating): Anker PowerCore Speed 20000 / 5-8V,3A / 8-10V,2.4A / 10-12V,2A

* **Other information** 
2 months ago the screen was nice and very bright, I left the TS80 in his box and now after plugging it to the same battery with the same cable the screen is barely readable.

JanaKo76, 2019-10-08:
Hallo, I have the same problem with this TS80 iron and have the same fw version as you. Can you please tell me if you are solved this issue ? Thanks.

Ralim, 2019-10-09:
Hi, this is not a firmware issue.

Screens going dark like this is most likely an underlying issue with the OLED display.
Usually this is fixed by replacing the display itself (either yourself or under warranty).

You can try contacting Miniware about this issue.

closed 2019-10-09 by Ralim