Ralim/IronOS #501: t80tris

BB-MBS, 2019-09-09:
* **I'm submitting a ...**
  - [x] Feature request

Please give to all users of your brilliant firmware an opportunity to astonish their friends and bosses. Give them the Tetris like on [TS-100](https://github.com/joric/ts100tris)
Just for fun :)

Ralim, 2019-10-09:
This has been bought up before, but adding it would require a fair amount of work as its based off a very old version of this firmware, and all of the code base has since been replaced.

It would also likely result in other features having to be removed to make room for the code to fit :(

discip, 2022-04-23:
@BB-MBS
Since @Ralim pointed out why this is not worth the effort, please close this issue. 😊

thanks

BB-MBS, 2022-04-23:
OK :)

closed 2022-04-23 by BB-MBS