Ralim/IronOS #502: Iron will not go to sleep.

Derek8563, 2019-09-11:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report possible bug

* **What is the current behavior?**
Iron will not go into power saving sleep mode no matter how long it's left on and undisturbed.

* **What is the expected behavior?**
Sleep after period set in settings.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

All I did was install V 2.05. It works fine when restored back to OEM firmware!
With the update installed you have to put iron to sleep manually by pressing the two buttons together.

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.05
  - PCB Version: (1/2): 1
  - Power Supply (Voltage and Current Rating) : 24V

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Neposedaua, 2019-09-21:
I have also a problem with the transition to sleep mode.
Verified on releases 2.05 and 2.06

ToXIc-Dev, 2019-09-27:
I am also having the same issue on V2.06.

Ralim, 2019-09-29:
This is a previously reported issue that appears to be affecting version 1 PCBs (the older accelerometer).

I'm currently away from my older unit so haven't been able test a fix yet. 😢

Saschko2000, 2019-09-29:
I have one that is affected. I could send it to you. But as I come from Germany it costs abaout 13€ one way, and I dont know if that makes sense?

Derek8563, 2019-09-29:
> This is a previously reported issue that appears to be affecting version 1 PCBs (the older accelerometer).
> 
> I'm currently away from my older unit so haven't been able test a fix yet. 😢

I'm a patient guy. I love the firmware so much I'll use it anyway and just upgrade as and when.

Thanks for the reply.

Ralim, 2019-10-08:
[TS100_temp_3fea.zip](https://github.com/Ralim/ts100/files/3702041/TS100_temp_3fea.zip)

Heya,
This is a build from my branch that I'm working on for some better temperature calibration/temp curve work.

Could you please use the debug menu to see if the move timer is working for you in this version?
Also, as this is from my temp branch, would love if i could crowd source some data from you all :grin: 
-> Please do a temperature calibration in the menu.
This is done by letting the tip and handle come into temp equilibrium (ie, leave on desk and ignore for an hour). Then power up the unit, and go into the Settings menu > Advanced > Temperature calibration > press the front button (for OK). Then it will dance " . . . . " (about 1.5 seconds).
Then show a checkbox and a 4 digit number.

Would be wonderful if you could post back your tip type and the 4 digit number. Hopefully most are around 700-900 range if my maths works out :D

NOTE:
-> This version may not have perfect temp calibration atm (though i hope its moderately close).
-> Should have some minor fixes for quality of life stuff like the power estimator too.
-> *Might* be better at temp control depending on your tip.
-> Built from git tag `3fea95c`

If you find the temp control is way off, let me know.

Neposedaua, 2019-10-14:
> Would be wonderful if you could post back your tip type and the 4 digit number. Hopefully most are around 700-900 range if my maths works out :D

My code is 0678
This firmware works well. 
Thank you for your work!

fricy, 2019-10-28:
Timer counts down with 3fea, and sleep mode triggers correctly on my end. I'm also not seeing the visual bugs from RC5, so I guess I don't need to report them. Only problem I have is with the auto orientation not working - the same as in 2.05.1.

Calibration code: 0696 / B2 tip

ZPolon, 2019-10-30:
У меня также есть та же проблема на V2.06-RC5. На V2.06-RC4 режим сна и автовыключение работает. Номер 79BA8AD7. DFU:3.45. Приобретен месяц назад.

flipmess, 2019-12-04:
> [TS100_temp_3fea.zip](https://github.com/Ralim/ts100/files/3702041/TS100_temp_3fea.zip)
>
> Heya,
> This is a build from my branch that I'm working on for some better temperature calibration/temp curve [...]

i can confirm that sleep works with TS100_temp_3fea. don't know what it has to do with temp control ^^
it wouldn't work with 2.05 or 2.06-rc5

Ralim, 2019-12-29:
This should be fixed in the latest release (I hope), please re-open if its not fixed and ill check again.

closed 2019-12-29 by Ralim

ToXIc-Dev, 2019-12-29:
This has fixed it for me, thank you very much for your work :)