Ralim/IronOS #503: TS80 Erratic Power Reading

limited660, 2019-09-22:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
Upon turning on, the power reading bounces all over the place during heating. Goes from 5w to 95w and everywhere in between, sometimes settles around 50-60w

* **What is the expected behavior?**
My understand is that this iron has an 18w max on 9v, why does the power reading behave as so?

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Install RC5
2. Turn on
3. Watch power reading

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

Issue does not seem to exist on RC4

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.6RC5
  - PCB Version: (1/2) - Unknown
  - Power Supply (Voltage and Current Rating) : Anker PowerCore Speed 20000 - 8-10v @ 2.4A

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-10-08:
This is a known bug, will have fix for this coming *soon*.

soon --> Fixed in my testing branch, but that branch is not ready for prime time or TS80 yet.

Ralim, 2019-12-29:
This should be fixed in the latest release (I hope), please re-open if its not fixed and ill check again.

closed 2019-12-29 by Ralim