Ralim/IronOS #504: Just a thank you!

Breakingen, 2019-09-25:
My TS80 arrived today but was buggy straight out of the box. It wouldn't work with any QuickCharge 3 power banks or AC adapters (also QC3 certified). The display said "Low Vot" for all power supplies.

I used a USB power meter to measure the power. It never went above 5.5V.

I upgraded from firmware 1.06 to official 1.07 but no fix.

I then flashed your unofficial firmware and it is now working. The voltage rises to 9V as soon as it is plugged in and it works exactly as I would expect.

Thank you so much!!!

whitehoose, 2019-09-26:
@Breakingen Good to see Ralim getting a little praise - people are very quick to pile on when there's a glitch, they sometimes forget all the good effort that's gone on round that.

I despair that the TS80 is so fussy about qc3.  It seems that manufacturers have reverse engineered the qc3 firmware so many ways now that even the same manufacturer may have mix of models that will or won't talk nicely.    
We had all this when the iron first appeared and we were all trying to power it with mixed results.
The banggood package was offered with a PSU that worked (and still does) the DN.OX it's offered separately too.  Fine if you have mains - but if you need portability Powerbanks were (are) a problem it seemed to be about 50.50 what worked and what didn't, despite being "certified" qc3
QC3 Adapters were about the same.

You'd need to wade through https://github.com/Ralim/ts100/issues/349 it's a long and rambling thread, but rather than entering the lottery at this late stage and wasting money, it might pay to spend a couple of hours seeing what was found to work

Vinigas, 2021-01-04:
#227

closed 2023-07-21 by Ralim