Ralim/IronOS #505: Unclear documentation on some things (temperature and voltage calibration)

epigramx, 2019-09-28:
On temperature calibration: If we change VIN, do we have to recalibrate? When it calibrates, does it assume the room has a specific temperature? If yes, which one? 

On voltage calibration: Is it required? What happens if one doesn't do it? It seems to be manually set only. Do we have to 'multimeter' our VIN and input it?

Ralim, 2019-09-29:
Changing VIN doesn't matter.

It's mostly actually calibrating the 3.3V rail and the slight variation in the tips.

Voltage calibration is generally not required. A small number of units have poor tolerance resistors where it can matter, but I would say for 99% of people it's not required.

epigramx, 2019-09-29:
And is temperature calibration always the same accuracy (if done correctly), or does it assume the room temperature is specific?

Ralim, 2019-10-21:
Should be same accuracy, so long as handle internal temp === room ambient temp.

epigramx, 2019-11-12:
^ does that imply the device has an independent temperature sensor? One in the device, one in the tip?

Sinclair-ZX81, 2021-03-13:
When is it best to run **Calibrate temperature** - immediately after powering on the iron (so the handle still is on room temp), or after some minutes, when the handle got standby operating temperature due to warm-up of internal electronics?

Ralim, 2021-03-13:
> immediately after powering on
Generally it should _not_ be required for most users

discip, 2022-04-23:
@epigramx
Please don't forget to close this issue. 😊

thanks

closed 2022-04-26 by epigramx