Ralim/IronOS #506: TS80 doesn't sleep correc

Bullone, 2019-10-05:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [ ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. 
2. 
3. 

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.x
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-10-07:
Hey,
Can you tell me which tip you are using?

This is the code that keeps power banks alive, putting too much power out.

Its intended to be slightly less than the power required to maintain the tip, so that the tip does cool down, but by pulsing small amounts of power it keeps power banks from going to sleep.

Ralim, 2019-10-21:
This is fixed somewhat in the testing power branch. Not ready for prime time yet, but a fix i coming :)

Bullone, 2019-10-21:
Great! Thank you for keep me aligned :)

Ralim, 2019-12-29:
This should be fixed in the latest release, re-open if its not fixed :)

closed 2019-12-29 by Ralim