Ralim/IronOS #509: ts100 bootup image not working

Mirrored6, 2019-10-08:
* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
bug

* **What is the current behavior?**
no getting bootup image on startup

* **What is the expected behavior?**
getting bootup image on startup

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.  i made a bootup image using the logo editor
2.  saved the .hex
3.  and uploaded it to the ts100 using the DFU mode.

* **What is the motivation / use case for changing the behavior?**
well i want a boot logo

* **What are you running:**

  - Firmware Version: 2.06
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating) : V = 19  A = 2.1
 
* here is the image and the .hex file
[.hex is uploaded on mega](https://mega.nz/#!mRA2XSaB!2PsRMB4f-CAxFWmh85kYWzmOWxzim8lsNkBtxNvhmKA)

[image]

* oh and is it possible to use an animation as bootup image???

saikerone, 2019-10-18:
Got the same Issue... No logo is showing on bootup

ggc161169, 2019-10-18:
I had that same problem, I resolved by installing version 2.04 (which had no problem with the boot logo) and then installed the following ascending versions

Mirrored6, 2019-10-18:
thank you for your answer ill try it tmrw.

Ralim, 2019-10-21:
Hi, Can you confirm what version of the logo editor you are using? there was an older broken version that some people have used...

ggc161169, 2019-10-21:
Hi, I don't manage well here in Github. The version I find for Win appears in https://github.com/Ralim/ts100/releases/tag/v2.02

saikerone, 2019-10-22:
version logo editor was ; https://github.com/Ralim/ts100/releases/tag/v1.15

saikerone, 2019-10-22:
> I had that same problem, I resolved by installing version 2.04 (which had no problem with the boot logo) and then installed the following ascending versions

When I go back to 2.04 ND FLash the logo, the screen goes black. When I reflash 2.04 its working again.

Ralim, 2019-10-22:
Generally that indicates you are using a bad logo generator version from before the bugfix.
Any chance you can try the python script? Its much better made...

saikerone, 2019-10-22:
ok i downloaded the logo editor from this version; https://github.com/Ralim/ts100/releases/tag/v2.02 
Also i changed my file from .jpg to .png. Then it worked for me. Thanks for your support on this.

Mirrored6, 2019-10-22:
ill try it out later. anyways thanks for the support. i am going to close the issue for now.

closed 2019-10-22 by Mirrored6