Ralim/IronOS #510: TS100 Sourcing Power for an LED

Eagle6705, 2019-10-12:
Not an issue but more of a question.
 I'm looking for 3.0 - 3.2 Volts to power a white LED  (5v will suffice I can use a resistor to drop)

My weller WPS18MP finally broke. Its my go to and 4th one mainly due to 2 features I can't find anywhere else... the length of the hot end and the LED which makes my soldering a breeze.
THE TS100 hot end is fine but the handle is a bit small for my taste but It is a great field iron.

I got a TS100 and designing an LED mount to go under right before the hot end.  I need to find a voltage source inside to power on the LED when power is provided to the ts100.  I'm looking at the schematic but I don't want to source from a spot that will effect the hot end or cause an issue that I am not aware of. 

(and yes I got a 5th weller to do the soldering )

:)

Ralim, 2019-10-21:
There is a ~3.3 rail on the pcb that is used for the STM32, that comes from a switching reg iirc.
You can probe for it on the caps on the small daughter board quite easily, or use the programming pads on the bottom of the board to tap it.

Eagle6705, 2019-11-08:
Strange..not sure why I didnt get an email you responded.... I'll keep you up to date I'll probably rework a ts100 grip to include an LED light, I'll see if I have time this weekend to probe for it

closed 2019-12-23 by Ralim