Ralim/IronOS #513: Crashing while reaching last information on ts100

TregliaRaphael, 2019-10-19:
* **I'm submitting a bug report**

* **Do you want to request a *feature* or report a *bug*?**
Just report a bug

* **What is the current behavior?**
When i'm on tips information in pause mode and i'm holding the button B, when i'm trying to reach the last value while pressing as many time as i can on A button, when i'm on TRES: 002, i can't do anything anymore. I'm forced to unplug the ts100 but evrything works fine again when i'm plugging it.

* **What is the expected behavior?**
I'll like to be able to don't be forced to unplug the ts100 to continue to use it

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Go on tips information (first view while powering on the ts100)
2. then Hold B to go on the ts100 information
3. Then press A until the TRES information
4. Now u can do anything anymore

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06
  - PCB Version: (2)
  - Power Supply (Voltage and Current Rating) : 23.5V

* **Other information** 

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2019-10-21:
This is an interesting bug.
I've refactored that menu a bit recently so quite possible its broke. 
When the power branch is more stable ill post another testing build :)

TregliaRaphael, 2019-10-21:
Yup, it's just that no one is exploring the ts 100 like this i guess ^^

Ralim, 2019-10-21:
Can you confirm if you were using a specific language of firmware?

TregliaRaphael, 2019-10-21:
It was the FR version

nikalexey, 2019-12-05:
Array index out of bounds 
const char* DebugMenu[] have from 0 to 12 items
but menu counter from 0 to 13
Crash when screen == 13
```
OLED::print(DebugMenu[screen]);
```

Ralim, 2019-12-24:
This should be fixed in the current master _I believe_.

closed 2019-12-26 by Ralim