Ralim/IronOS #514: Displayed input voltage doesn't change

RChadwick7, 2019-10-20:
I don't know if this is only cosmetic, or if the software makes decisions based on the input voltage. I have one of those Ebay  QC3 PCB's that will connect to USB PD, and tell it to put out different voltages. It has a pushbutton to change voltages, but defaults to the lowest. When I connect it to my TS100, the TS100 displays somewhere around 12V, but when I push the button so that it's output is 19V, the TS100 still says 12V. It looks like the TS100 only measures input voltage once at startup.

Ralim, 2019-10-21:
Hi,
Can you confirm what firmware build you are on?
On some older firmwares there was a bug where it didnt update the voltage array at startup quickly.
Have you validated that you are actually getting 19V with a multimeter?

RChadwick7, 2019-10-21:
When I first encountered this, I didn't know what version I had, so I downloaded and installed the latest. While I did not verify with a multimeter, if I switch the module to 19V, and then plug the TS-100 in, it shows 19.1 or so. If I leave it plugged in, then switch, it stays low. Also, this new firmware won't power up on the module's first setting, but does on the 2nd or 3rd. Whenever I switch to a mode that will power the TS-100, the input voltage stays there and doesn't change.

Ralim, 2019-12-29:
This should be fixed in the latest release, would love it if you could check :)

RChadwick7, 2019-12-29:
It does appear to be fixed, although it is slow to update the voltage. I'm not sure if it's a feature or not, but climbing from 14 to 19 volts takes about 5 seconds. It doesn't suddenly change, it climbs. I can certainly live with it like that :)

Ralim, 2019-12-31:
Yeah its a little bit slow, so that when the tip suddenly starts heating it doesn't trip the under voltage straight away.

Sounds like its working better now, so going to close this off (but feel free to re-open this if desired).

closed 2019-12-31 by Ralim