Ralim/IronOS #516: Missing last digit in voltage adjustment menu.

discip, 2019-10-22:
Good evening,

* **I'm submitting a ...**

  - [x] Bug report (I really do not know, if this was intended, so I assume it got lost by accident.)

* **What is the current behavior?**

In the voltage adjustment menu the last digit is not displayed.
(At first I thought it would not change at all, until I realized it changes by a hundredth of a Volt.)

* **What is the expected behavior?**

Would be nice, if the last digit would be displayed.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

Power on your device, press the B-button until the Advanced Options screen is reached. Now press the A-button, than press the B-button again, until 'Calibrate input voltage?' is displayed. At this point press the A-button. After that you can either press the A-button do raise the voltage or the B-button to lower the voltage by 0.01V.

* **What is the motivation / use case for changing the behavior?**

The convenience of not having to count the presses to reach the desired value.

* **What are you running:**

  - Firmware Version: v2.06-RC5
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating):

voltage | amperage
-|-
3.6 - 6.5V |   3A
6.5 -   9V |   2A
9   -  12V | 1.5A

Thank you in advance
kind regards

discip, 2019-10-23:
Thank you for the explanation.
Is this behavior going to stay like this, or are there any plans to change that, so that the voltage is changed by 0.1V increments per press and displayed accordingly?

Ralim, 2019-10-23:
There is no bug in how this is operating, as the entire point of the calibration is to create  a method to convert ADC to the voltage.  More precisely, its a chicken and the egg problem, to be able to allow for adjustments like your asking (0.1V step), we need to know the calibration.... Which is what this does.

So i have no plans to change this at the moment :sweat:

closed 2019-12-28 by discip