Ralim/IronOS #517: remove extra 0 (zero) in front of one-digit numbers

discip, 2019-10-22:
Good evening,

* **I'm submitting a ...**
  - [x] kind of Feature request

* **Do you want to request a *feature* or report a *bug*?**

I want to request a little change in the appearance of the interface.

* **What is the current behavior?**

The current way of displaying one-digit numbers is, to show a 0 in front of it.

* **What is the expected behavior?**

In my opinion it would be great, if the the one-digit values would be shown just as such.

* **What is the motivation / use case for changing the behavior?**

It might be a touch of perfectionism.

* **What are you running:**

  - Firmware Version: v2.06-RC5
  - PCB Version: 2
  - Power Supply (Voltage and Current Rating):

voltage | amperage
-|-
3.6 - 6.5V |   3A
6.5 -   9V |   2A
9   -  12V | 1.5A

Thank you in advance
kind regards

Ralim, 2019-10-23:
Hey,

To clarify, would you want this in only certain locations or everywhere?
Would you want the numbers to be left or right aligned (currently numbers right align)?

discip, 2019-10-23:
Good afternoon,
thank you for the instant response.
I think, it would be best, if the appearance of numbers would be system wide the same. And if I am allowed to choose, I would prefer to have the numbers aligned right.

Ralim, 2019-12-23:
@discip 
I'll try and get this thrown in shortly, as its a trivial patch to do.
Also, 😍  for putting the charger rating in a nice table, while not required for this issue, I appreciate it.

JohnEdwa, 2019-12-23:
Though seeing leading zeroes shows the user the scale of the value in question, if I see "01" I know the maximum value is at least 10, while a "1" in a system without leading zeroes doesn't tell me anything.

Which does bring me to a suggestion, would it be possible to somehow mark the last value before rolling over?   
Taking the Boost setting as an example, right now it goes "430 >440 > 450 > 250". Would it be possible to bold or invert or somehow mark the "450" to indicate that it's the highest it can go, pressing again rolls back to the lowest value.

Ralim, 2019-12-28:
I believe this is now done by #531 and #534

Ralim, 2019-12-28:
@JohnEdwa 
> Which does bring me to a suggestion, would it be possible to somehow mark the last value before rolling over?

Probably can be, can you spawn this out to its own issue please :)

closed 2019-12-28 by Ralim

discip, 2019-12-28:
@Ralim 
Thanks for finally releasing the long awaited built.
ps: I like it sorted.

Ralim, 2019-12-29:
> ps: I like it sorted.
Its purely as sorted as my file browser when i drag and drop, so hopefully it stays sorted :|