Ralim/IronOS #518: Problem screen OLED TS100

trunglp, 2019-10-24:
Hi everyone,

Today i problem with OLED screen TS100. I don't see text on screen. how to fix it or replace screeen

[image]

Ralim, 2019-10-24:
Hi,
It looks like you have screen burn in on your OLED.
This is not the manufacturers support forum, so I don't have magic to offer.
The screen is a moderately common part that can be picked up from Aliexpress, but you can also reach out the manufacturer for information / replacement parts etc.

chaoticchoas, 2019-11-12:
Hello I had the same issue though in my case I haven't used it in months since I only solder occasionally.

jypma, 2019-11-23:
I have the same problem. I'd be OK to switch out the screen myself... could someone point to the "common part" on AliExpress? A quick search seems to point out several varieties of similar size...

nreckart, 2020-01-09:
I'm having the same issue. Screen worked just fine months ago. Put the iron away in a case for several months. Pulled it out a couple nights ago to do some soldering and the screen is almost completely unreadable. How does screen burn occur when the iron isn't even being used? 

What replacement display could be ordered/used? Cheers!

trunglp, 2020-01-10:
> I'm having the same issue. Screen worked just fine months ago. Put the iron away in a case for several months. Pulled it out a couple nights ago to do some soldering and the screen is almost [...]

I replace my OLED, and it work very well, you can buy it from https://s.click.aliexpress.com/e/NSYo8q9w

0.69 inch OLED display module 14PIN UG-9616TSWCG02 96*16

nreckart, 2020-01-10:
Thanks @trunglp! Much appreciated!

simontom, 2020-04-24:
It seems this is pretty common issue. :-(

jypma, 2020-04-24:
@simontom I was successful in replacing mine, I wrote a little bit about over at [#477](https://github.com/Ralim/ts100/issues/477#issuecomment-580196408).

xamindar, 2020-07-09:
Someone on a youtube post about this issue said it is caused by this Ralim firmware. Mine has this issue as well. I only used my TS-100 a couple of times a year ago and just pulled it out again and the screen is like this. 

Can anyone confirm this issue is occurring with stock firmware as well?

Ralim, 2020-07-09:
It has been reported on the miniware firmware as well.

I'm using the same oled screen initialisation code as miniware so it's not like I'm driving it "harder".

digitalis-ts100, 2020-07-29:
Hiho,
It happend to me too.
I upgraded the firmware (2.7.)  ralim  when i first got my ts100 July 7,2019.  And all seem ok. since i not do much soldering i stored the iron for months.  then time comes i want to use it again and noticed the lcd started to blur. then it disappears. 
It took me a while to find a replacement 0.69 oled on lazada. finally, 3 weeks ago i found a seller and today it arrived and i changed it and it's working great again.
I dont think the **ralim firmware** can cause the **oled to die**. i bought 2 so i still have the option to replace it again if it does die again. **and might put back the original firmware if it happens again.**
it would be nice if someone with original firmware with oled gone bad let us know 👍 so we can isolate and not blame ralim's firmware.

regards

xamindar, 2020-08-10:
Yeah you are probably right. Doubt it's Ralim firmware. Thanks for responding Ralim. 
I replaced my screen as well so we shall see. Hopefully it lasts this time. I wonder how common this issue actually is.

Griffsnog, 2020-09-18:
> Hiho,
> It happend to me too.
> I upgraded the firmware (2.7.) ralim when i first got my ts100 July 7,2019. And all seem ok. since i not do much soldering i stored the iron for months. then time comes i [...]

I've had mine for almost 2 years. I don't know how to change the firmware or why i would.  This happened to mine last week.. I solder pretty regularly. At least twice a week but have been using it less the past few months.  My screen sometimes comes on at full brightness for a few seconds like nothing is wrong.  I cannot get this to reliably repeat by shaking it or anything.  I have only ever used a power source other than the one it came with a couple of times and the one i used was far weaker than what it came with.

jypma, 2020-09-21:
I wonder, for the people where the iron failed, was it connected to power during idle? Or did you completely remove the power when it was not in use? Since my screen failed, I've been disconnecting it from power when not using, to rule out any kind of idle current affecting this.

OpenBox3D, 2020-12-11:
https://www.youtube.com/watch?v=mBlsF0Q6YOQ

catch 22, you need a working soldering iron to replace the display of the faulty soldering iron display

dzarda, 2021-02-08:
Could this be related to storage temperature?

Ralim, 2021-02-11:
@dzarda 
Absolutely, OLED screens degrade depending on their environment as they have a finite lifespan.

DocSneider, 2021-03-25:
For those who are looking for a replacement.

I just replaced mine with that one ["Graphic I2C 0.69 inch 96x16 OLED Display MP3 Module,White on Black"](https://www.buydisplay.com/graphic-i2c-0-69-inch-96x16-oled-display-mp3-module-white-on-black), from www.buydisplay.com.

It works quite well and the display looks fine.

Houseman74, 2021-05-01:
Hi. Does the provided OLED Display 96x16 replacement fit also for the TS80?
Mine is slowly dying too..
Best regards

Ralim, 2021-05-01:
@Houseman74 It should work in the Pinecil/TS100/TS80/TS80P as far as I understand.

Houseman74, 2021-05-02:
Thank You Ralim

rpc72, 2022-02-22:
> Hiho, It happend to me too. I upgraded the firmware (2.7.) ralim when i first got my ts100 July 7,2019. And all seem ok. since i not do much soldering i stored the iron for months. then time comes i [...]

Mine was on the original Miniware firmware when it died.  Bought it off Lazada as well.

discip, 2022-04-23:
@trunglp
This is not software related and obviously fixed.
So please close this issue. 😊

thanks

simontom, 2022-04-26:
IMHO, lowering the backlight would not help at all.
If you read the posts (well, not sure if it was written in this issue too) you can see people have problems even tho' they don't use it for some period at all.
I have the same expirience, didn't use it for like months and the display was totaly screwed. I was so disappointed.

tel001, 2022-08-27:
My TS-100 was in a pencil box on the shelf unconnected and it still developed the display failure! I put the Ralim firmware on it right after buying it. It was working as usual when I last put it away and did not use it for about 6 months as I had just bought a new soldering station that became me daily use iron. I needed the portable TS-100 for a small job and it LET ME DOWN HARD! At least I have a micro soldering iron to fix it.

closed 2023-07-21 by Ralim

antimix, 2025-02-01:
It happened to me to. I am using the original firmware, and I don't think it is related to the firmware.
Also in my case, after some months, suddenly it was dimmed.
I have two ideas on why it happens:

1.  Soldering at high temp for long time (e.g. 360-380°) trying to remove industrial high temp tin.

When this happens may be that the OLED screen catch too much heat and it get damaged. 
However I normally use a professional de-solder gun air station and not the TS100, so I don't see this cause as probable.

2.  Too much power sent to the OLED. 

May be a case, but the issue happened just few week after I purchased a big 24V supply power for the TS100. 
It lasted for years when I used it powered by my old 12V power supply. But I wanted much power on the tip, for fast speed and large surfaces, and so I got a 24V  power supply. But few week after it failed 😪. 
So I suspect that the TS100 circuit board may send to many mA to the OLED when the TS100 is receiving 24V or is using high current.

Today I replaced the OLED.  I suggest all you to not use normal tin to solder the new OLED connector, since it can be melted or damaged. I used a special LOW TEMPERATURE tin, that melts at 138°, so I did not risked to damage the thin plastic connector.