Ralim/IronOS #521: TS100 stock firmware works, this one results in a black screen

HammerPavel, 2019-11-10:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [X] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
official firmware - behaves normally, screen works, tip works, minor artifacting in some menus

open source firmware - screen in DFU works normally, DFU itself works normally, just connecting usb doesn't do anything(as expected), powering on screen remains dark. sometimes when plugging in screen flashes a couple of lines of pixels. i do not know the menu structure of the firmware but i tried buttonmashing the tip doesn't start heating up

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.???
2.
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.18 official; tried 2.02 2.03 floss
  - DFU version 3.45
  - PCB Version: (1/2) v2.5(? see image) bought last month
  - Power Supply (Voltage and Current Rating) :19v 65w

* **Other information**
firmware loading DOES show .RDY for both versions of floss firmware and for the official firmware

HammerPavel, 2019-11-10:
board shot

HammerPavel, 2019-11-10:
board shot: https://files.catbox.moe/5ehw17.jpg
github uploading doesn't work rn for some reason

Ralim, 2019-11-19:
Interesting, 
It's been ages since these issues last showed up.
One big dfference between this firmware and the stock firmware is that it checks error codes etc to make sure stuff worked more. Not sure if you are running into screen issues or something else holding up the boot

HammerPavel, 2019-11-19:
can i compile with a flag to disable those just for a test?

Ralim, 2019-11-23:
Not really, Most of the error checking is part of the ST HAL, and otherwise It feels like a bad idea to blindly spew out data onto the I2C bus if there is something wrong with a part on the bus.

As you have a rev2 unit, could you try checking that the daughter board is mounted properly, inspect solder joints etc.

Failing Accelerometer will only hold up boot if its messing with the I2C bus. Failing screen can hold up boot as its assumed that a TS100 / TS80 will have a behaving screen.

HammerPavel, 2019-12-22:
i didn't get anywhere with investigations, as it appears this is the only ts100 to do this, lets call this issue a random hardware deffect

closed 2019-12-22 by HammerPavel

fastriker, 2020-02-11:
I have the same issue. DFU 3.45, PCB version I don't know, because I did not opened it now.
Original firmware works fine. Current firmware from Ralim results in a black screen.
I flash under Ubuntu Budgie 18.04.