Ralim/IronOS #522: Power pulses for TS 80 do not work

thomasgi1, 2019-11-19:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x ] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
When the TS80 entered the sleep mode the power bank (Charmast Powerbank 20800mAh) turns off after 30s delay.

* **What is the expected behavior?**
The power pulses keep the power bank alive also in sleep mode.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Plugin the TS80 into the power bank
2. Turn The TS80 on
3. Wait until sleep mode is entered (1 minute)
4. Wait additional 30 seconds
5. The power bank turns off

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
Currently the sleep mode can't be used with a power bank.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06-RC5
  - PCB Version: (1/2) ?
  - Power Supply (Voltage and Current Rating) : Charmast Powerbank 20800mAh / 9V 

* **Other information**
The code in startPIDTask() is not going to work regarding the power pulses. Power pulses are only generated, if the currentlyActiveTemperatureTarget is equal to 0. But in case of the sleep mode the currentlyActiveTemperatureTarget is the sleep temperature. IMHO the power pulses should be generated independently of the temperature target. Instead they should be generated, when the tipMilliwatts where 0 for a longer time (say 10 seconds).

mihalski, 2020-01-06:
I have a similar problem with the Romoss 20+ power bank and the 2.07 firmware. 55 seconds after I plug the soldering iron in (and without any heating) it turns off. Shortly after it stops heating the same thing happens, but I haven't measured that interval.

Ralim, 2020-01-07:
@mihalski 
The original issue was fixed, the TS80 now draws power pulses every few seconds. However not all power banks will work with that, some require a higher average.

It could be added to the menu, but too much more and you will forever be heating the tip, which causes issues too.

whitehoose, 2020-01-07:
The sleepy powerbrick just seems to stick like monkey snot on a blanket - A quick fix is to connect A LED to the second outlet if you only have one outlet you can easily make a short plug and socket extension with a LED.  You could get fancy with a mosfet switch or just a variable resistor.  My powerbank has a bypass but my main source is 4s Lipo and AOKoda adapter which never sleeps.  It's not clever but It seems that the QC3 bunny has an infinite number of gotchas.
Sometimes a tap with a hammer is the best solution in the global scheme of things - A simple LED is unlikely to conflict - unlike the more sophisticated solution seems to do

closed 2020-04-04 by Ralim