Ralim/IronOS #525: Firmware performance issues

akamensky, 2019-12-03:
* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **What is the current behavior?**
Temperature overshoots by a whopping 50C and does not seem to hold steady, it without any load applied dips to about 30 below configured before heating turned on again.

12V option selected, but only 9V drawn (with reliable QC3 PSU). (in heating in UI shows digit 9 on the left, not sure if that is implied to be a voltage?)

* **What is the expected behavior?**
Temperature doesn't overshoot configured one, heating applied earlier (don't let it dip too low, or at least provide configuration option at what gap the heating should be turned on).

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**
Just using latest released version.

***Steps to reproduce:***

1. Upgrade to latest released version (as of date of this report)
2. Use it
3. Configure 12V

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: v2.06-RC5
  - PCB Version: 
  - Power Supply (Voltage and Current Rating): 24V/5A

Ralim, 2019-12-23:
Hi,
Have been working on a re-worked temp control system and calibration system for a while (got interrupted by life). 

>Temperature overshoots by a whopping 50C and does not seem to hold steady, it without any load applied dips to about 30 below configured before heating turned on again.

Could you try the attached file on your iron to see if the temp control is better behaved for you? (flash the TS80 file).

If you could also do a temp calibration, and report back the calibration number, would be really handy. Need to figure out a rough good default before a release goes up.

To do a temp calibration:
Leave iron to sit so that iron handle and tip are the same temp.
Then plug in, and goto the advanced menu -> temp calibration -> Press front button -> Wait for dots -> record the number it shows you.

[2.06_temp_TS80_TS100_EN.zip](https://github.com/Ralim/ts100/files/3994852/2.06_temp_TS80_TS100_EN.zip)

>12V option selected, but only 9V drawn (with reliable QC3 PSU). (in heating in UI shows digit 9 on the left, not sure if that is implied to be a voltage?)

What make / model is the QC power supply that your using? (24V/5A doesn't make sense for QC3). Have you validated somehow that the power supply actually supports 12V mode ?

onovy, 2019-12-24:
I have/had same problem as @akamensky 

Ad temp: Tried to flash today's master branch a temperature overshoot is fixed for me
Ad 12V: I tried ~5 chargers and only one of them works on 12 V (Xiaomi). Other ones are Blitzwolf and Nokia original one. All of them are QC3.0 and supports 12 V.

onovy, 2019-12-26:
Whole day work with my TS80 with master branch firmware and it's rock solid. Thanks!

closed 2020-04-04 by Ralim