Ralim/IronOS #526: Master is broken after merging 3f69dbd5

joric, 2019-12-16:
Master is supposed to build fine as this is the official master branch but there are errors:

1) NameError: name 'getDebugMenuHeaders' is not defined
2) cmsis_os.h:465:76: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings] { #name, (thread), (priority), (instances), (stacksz), (buffer), (control) }

I guess the recent merge from #508, particularly commit 3f69dbd5 got merged wrong, there is textList.extend(getDebugMenuHeaders()) but no function declaration.

As for the second issue it's probably bad translation file but just for the record I'm using arm-none-eabi 6.3.1.

Firebie, 2019-12-17:
Fix:
diff --git i/Translation Editor/make_translation.py w/Translation Editor/make_translation.py
index 803dfb1..8226ac1 100755
--- i/Translation Editor/make_translation.py	
+++ w/Translation Editor/make_translation.py	
@@ -176,7 +176,6 @@ def getLetterCounts(defs, lang):
     constants = getConstants()  
     for x in constants:
         textList.append(x[1])
-    textList.extend(getDebugMenuHeaders())
     textList.extend(getTipModelEnumTS100())
     textList.extend(getTipModelEnumTS80())  
     textList.extend(getDebugMenu())

joric, 2019-12-17:
Nope, not a fix.

Firebie, 2019-12-18:
I have used 9-2019-q4-major for win32 and I don't have second issue

nikalexey, 2019-12-18:
I don't have second issue too.
ArchLinux (Linux 5.4.3-arch1-1)
arm-none-eabi-gcc (Arch Repository) 9.2.0

joric, 2019-12-18:
Well it compiles, but there's still a wall of warnings. I'm not sure why and how to fix that. I guess getDebugMenuHeaders body still needs to be properly merged with all the fixes and defines.
```
Building firmware for TS80 in EN                                                                                         
./Core/Src/GUIThread.cpp:710:5: warning: "ACCELDEBUG" is not defined [-Wundef]                                           
 #if ACCELDEBUG                                                                                                          
     ^~~~~~~~~~                                                                                                          
./Core/Src/main.cpp:244:5: warning: "ACCELDEBUG" is not defined [-Wundef]                                                
 #if ACCELDEBUG                                                                                                          
     ^~~~~~~~~~                                                                                                          
In file included from ./Core/Inc/FRToSI2C.hpp:11:0,                                                                      
                 from ./Core/Inc/MMA8652FC.hpp:12,                                                                       
                 from ./Core/Src/main.cpp:2:                                                                             
./Core/Src/main.cpp: In function 'int main()':                                                                           
./Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.h:465:76: warning: ISO C++ forbids converting a string cons
tant to 'char*' [-Wwrite-strings]                                                                                        
 { #name, (thread), (priority), (instances), (stacksz), (buffer), (control) }                                            
                                                                            ^                                            
./Core/Src/main.cpp:82:2: note: in expansion of macro 'osThreadStaticDef'                                                
  osThreadStaticDef(GUITask, startGUITask, osPriorityBelowNormal, 0,                                                     
  ^~~~~~~~~~~~~~~~~                                                                                                      
./Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.h:465:76: warning: ISO C++ forbids converting a string cons
tant to 'char*' [-Wwrite-strings]                                                                                        
 { #name, (thread), (priority), (instances), (stacksz), (buffer), (control) }                                            
                                                                            ^                                            
./Core/Src/main.cpp:87:2: note: in expansion of macro 'osThreadStaticDef'                                                
  osThreadStaticDef(PIDTask, startPIDTask, osPriorityRealtime, 0,                                                        
  ^~~~~~~~~~~~~~~~~                                                                                                      
./Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.h:465:76: warning: ISO C++ forbids converting a string cons
tant to 'char*' [-Wwrite-strings]                                                                                        
 { #name, (thread), (priority), (instances), (stacksz), (buffer), (control) }                                            
                                                                            ^                                            
./Core/Src/main.cpp:92:3: note: in expansion of macro 'osThreadStaticDef'                                                
   osThreadStaticDef(MOVTask, startMOVTask, osPriorityNormal, 0,                                                         
   ^~~~~~~~~~~~~~~~~                                                                                                     
```

Ralim, 2019-12-23:
Should have fixed these now.
My bad for leaving master broken for so long.

closed 2019-12-26 by Ralim