Ralim/IronOS #537: Problems with my power supply since the latest two firmwares (and some other little things)

TheUntouchable, 2019-12-28:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Reporting a bug

* **What is the current behavior?**
My Charmast 20800 mAh QC 3.0 was working without problems but since the latest and the firmware before that it will not switch to 12V but stay at 9V and also it will be switched off after some seconds of idle.. Also the following zero will not be shown on the volt display but the dot ;) And if you don't have the tip inserted there is no error message but a temp of 430 shown.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06
  - PCB Version: (1/2) ? TS80
  - Power Supply (Voltage and Current Rating) : Charmast 20800 mAh QC 3.0

Ralim, 2019-12-29:
Hi,
Okay, so unpacking this:

> will not switch to 12V but stay at 9V

Okay, so your power supply must not support QC2.0 12V but only QC3.0.
Probably going to have to add a settings menu option around this.

> it will be switched off after some seconds of idle

Yeah, this is known, that feature was originally dropped as it only worked for a few people, but it will probably come back in next release.

> the following zero will not be shown on the volt display but the dot 

Which Voltage display? (How can i replicate this)

> And if you don't have the tip inserted there is no error message but a temp of 430 shown.

Not sure what to do on this one, as its actually an awkward thing to detect and appears to vary unit to unit. Open to ideas on the best way to detect this... At the moment, sorta inclined to go with the logic that users will probably notice a missing tip fairly quickly.

TheUntouchable, 2019-12-29:
Hello Ralim,

Thank you for your quick reply!

I am looking forward to the changes that will make my Powerbank work again as I was used too ;) If you need help with testing here, please let me know!

Here is the voltmeter I was referring to, maybe that's also caused by the German language:
[image]

Has the stock firmeware not a tip check already build in? But youre right, without a tip soldering would be very complicated :D

Ralim, 2019-12-30:
> I am looking forward to the changes that will make my Powerbank work again as I was used too ;) If you need help with testing here, please let me know!

I'll let you know, Some day I need to figure out a cheap way to get a lot of QC devices for better testing than my current 3 :grin: 

> Here is the voltmeter I was referring to, maybe that's also caused by the German language:

Thanks, I'll have a dig when i get time.

> Has the stock firmware not a tip check already build in? But youre right, without a tip soldering would be very complicated :D

Probably does now, I was the first to have it :grinning:. The current method looks for a reading in the top ~ 0.5% of the ADC's range, but not all irons will read that high due to unit-unit variations. Could probably add a sanity check for temp change rate, but I wonder how useful it would be.

thierer, 2019-12-30:
> > will not switch to 12V but stay at 9V

I just put v2.06 on my TS80 and seem to have the same problem. (Also, I'm confused: I'm talking about the setting labeled "power source" which has the options "9V" and "12V" though the help text for that setting refers to "Power wattage". We're really talking "voltage" not "wattage" here, right?)

> Okay, so your power supply must not support QC2.0 12V but only QC3.0.

I'm not sure if I understand correctly but I think you mean "That behaviour is to be expected if the power adapter does not support 12V via QC2.0 (but only using QC3.0)". If so, that's not the case with my power adapter: I know for sure that it supports 12V using QC2.0.

Please note that v2.06 is the first version I ever used so I can't say if it's a regression or not. If it helps I would be happy to try an older firmware, though.

anpage, 2019-12-30:
I'm having a similar experience. I just unboxed a new TS80. It didn't work with my QC 3.0 [Ravpower RP-PB172](https://www.ravpower.com/products/rp-pb172-portable-charger-20000mah) battery bank on stock firmware, so I flashed the final v2.06. It will negotiate 9v just fine now, but I have yet to get 12v working.

The same story goes for a [Xinspower Q183](http://www.xinspower.com/intelligent/product/28.html) QC 3.0 wall adapter, as well as an [Aukey PB-T3](https://www.aukey.com/products/16000mah-power-bank-with-quick-charge-2-0) battery bank, and [Aukey CC-T5](https://www.aukey.com/products/car-charger-with-quick-charge-2-0-cc-t5)/[Aukey CC-T6](https://www.aukey.com/products/dual-port-car-charger-with-quick-charge-2-0) car adapters, all QC 2.0. The only difference being that the 2.0 supplies will reach about 9.2v

Granted, the max power output of all 5 supplies is rated at 18W, so I don't think I'd benefit from 12v anyway, but I thought it might help to report on compatibility.

Ralim, 2019-12-30:
Thanks for the report, fairly sure I know the issue.
Will try and get a fix out _soon_

Ralim, 2019-12-31:
#539 and #540 should address these two issues.
Probably going to do a beta build _soon_

thierer, 2019-12-31:
Just checked with 0402b7bd and can confirm that my TS80 now uses 12V when set to that voltage. Thanks for the quick fix!

It's drawing 2A, though, which is a bit much for my power adapter and makes it drop to ~10V, but [I guess that is to be expected](https://github.com/Ralim/ts100/blob/0402b7bd2f7bc29c8040837566005a8f3d9ca1bb/workspace/TS100/Core/Src/hardware.cpp#L292)?

Ralim, 2019-12-31:
Yep, that is 1000% expected at the moment.

We cant really limit the peak current, as the tip is ~4.5 ohms, V=IR, 12/4.5=2.6A peak.

One thing next on the horizon is to bring in a nice menu to let you limit the average power to a certain wattage.

anpage, 2019-12-31:
The new release works very well for me. There are still some quirks, but I wouldn't be surprised to learn that it's because of the supplies and not the iron:

- Ravpower RP-PB172 battery bank (QC 3.0)
  - 9V setting negotiates 9V correctly
  - 12V setting negotiates 12V and causes the iron to presumably trip the over-current protection. The bank shuts itself down.
- Xinspower Q183 wall adapter (QC 3.0)
  - 9V setting negotiates 9V correctly
  - 12V setting negotiates 12V and draws almost 30W according to the iron's display. It seems to work fine regardless.
- Aukey PB-T3 battery bank (QC 2.0)
  - 9V setting negotiates 12V and draws about 26W when heating. This supply in particular doesn't seem to respect the request for 9V with v2.07, but it _was_ being limited to 9V with v2.06.
  - 12V setting: same as 9V
- Aukey CC-T5/CC-T6 car adapters (QC 2.0)
  - 9V setting negotiates 9V, runs from 8.8V to 9V when heating.
  - 12V setting presumably negotiates the highest possible voltage, but it remains at a maximum of about 9.2V

Thanks for the firmware and the prompt fixes. I'm loving this iron so far.

TheUntouchable, 2020-01-01:
Thanks a lot Ralim, the voltage issues seems to be fixed!
Funnily enough, my Powerbank now switches off its display when the soldering iron is in idle. But it does not switch off completely :D
Just a little hint, but the 0 is still missing after the dot at the volt display ;)

whitehoose, 2020-01-02:
I do have a QC3 power brick which works but only goes to 9v HOWEVER while I have a number of USB2 bricks I bypassed the expense and uncertainty of USB3 bricks (the lottery of will it won't it be compatible, or fall asleep or .....explode!!!) and went the Lipo route instead.
I can confirm the AOKoda Lipo to USB Power Converter QC3.0 Adapter works with the TS80 iron. as well as any phone I've tried it with as does BatteryGo BG-8s as well as some generic unbranded adapters bought on e-bay.
If you convert from powerbricks to Lipo batteries you can power a TS100 directly from s4 upwards or via the AOKoda adapter you can work a ts80 or any other USB 2 or 3 device up to and including qc3 @12v.
Using a decent Buck Boost converter you have enough power available to run a wide range of devices.  Because USB powerbricks use either boost or a combo buck boost to convert LiLon's 3.8v to 5v or 5>9>12v you are more likely to get a stable voltage from a lipo because you're only transforming it once.   

Price-wise they compare well and from a compatibility angle you can choose from quite a range of QC3 adaptors the AOKoda and BatteryGo BG-8s both use a widely compatible version of QC3 for certain.  I have other generic ones which also work well too - Ironically more cheap adaptors seem to be QC3 compatible than expensive power bricks.

Once you know yours is - you can plug any brand of lipo into it and be certain it will work across the board.  On the downside - lipos are chunky little beggars - and you need a dedicated charging method - but that said, you only need one. which then will fit all of them.

My bigger (biggest) worry is over driving my potentially delicate TS80.  Left to 9v, mine took 15 seconds (ish) to hit 320c and quickly 15seconds (ish) made a decent puddle on a 20g slug of copper I use for testing.  I don't see 12v making significant improvements on that - I do see overloading the Iron and tip potentially destructive.  An interesting option - but I'm not convinced It's worth burning out a tip or iron for.

closed 2020-09-14 by Ralim