Ralim/IronOS #543: TS80 2.07 (2.08?) missing trailing zero(s)

bradanlane, 2019-12-31:
Note: this refers to the **second** release of 2.07 with a date of `31-12-19`

Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [*] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
idle detail view example:
```
Tip: 28 Set:220
Input V: 12.
```

* **What is the expected behavior?**
```
Tip: 28 Set:220
Input V: 12.0
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. load firmware v2.07 (31-12-19) on TS80
2. set idle screen to "detailed"
3. view idle screen

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
When the voltage ends in anything other than a zero, eg `11.4` it shows the decimal digit but when the voltage is a whole number, it ends with the decimal point and no trailing zero.

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.07 (31-12-19)
  - PCB Version: (1/2)
  - Power Supply (Voltage and Current Rating) :

* **Other information**
I suspect the version number sound also have been changed to 2.08

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

cam-movandi, 2020-01-12:
In TS100 2.07 as well, probably related:

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
```
Power: 5.   55S
Tip:360C 20. V
```

* **What is the expected behavior?**
```
Power: 5.0   55S
Tip:360C 20.0 V
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Set soldering detailed mode
2. Have a power or input voltage with a trailing *.0 W/V
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
Release notes mentioned this was corrected, couldn't see any change in behaviour

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.07
  - PCB Version: (1/2)? TS100 early model
  - Power Supply (Voltage and Current Rating) : 5S li-ion tool battery

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

reversebias, 2020-01-12:
In TS100 2.07 as well, probably related:

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Bug

* **What is the current behavior?**
```
Power: 5. W  55S
Tip:360C 20. V
```

* **What is the expected behavior?**
```
Power: 5.0W   55S
Tip:360C 20.0 V
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Set soldering detailed mode
2. Have a power or input voltage with a trailing *.0 W/V
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**
Release notes mentioned this was corrected, couldn't see any change in behaviour

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.07
  - PCB Version: (1/2)? TS100 early model
  - Power Supply (Voltage and Current Rating) : 5S li-ion tool battery

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

Ralim, 2020-12-29:
Closing this as it has been fixed at _some point_  a while back 😰

closed 2020-12-29 by Ralim