Ralim/IronOS #544: Hint: use normal windows copy - using "Copy Handler" results in ERR file when updating

bl4k1st, 2020-01-01:
A problem I stumbled upon:
I had trouble updating my TS100.
I always got a ERR file.

Reason was I am using "Copy Handler". Copying the file with normal windows copy worked, the iron updated and I got a RDY file.

Perhaps you want to edit that in Readme install notes.

whitehoose, 2020-01-01:
Don't know if I'm misunderstanding I normally use linux - so rarely update from windows - but for me - a "normal" update sequence using either windows or linux is: Plug Iron in (holding button a) - file manager window opens DFU shows on iron - drop update into window ... window closes then opens displaying .err file .... ignore drop update file again - window closes pause .... then opens.
Recycle iron 
Check update OK
Its done this from day one. Linux NEVER shows a .rdy - windows tends to - but an impatient restart doesn't but still does the job.
because there is so little system space for frills - the messages are as misleading as the early DOS or windows errors.
The important thing is getting the new version number after second restart.

Ralim, 2020-01-08:
@bl4k1st There is a long discussion github issue about this (well many issues that get spawned).
It is also noted over in the [wiki](https://github.com/Ralim/ts100/wiki/Upgrading-Firmware) that using file copiers such as teracopy will break things.

I will try and add it to the wiki as well

PR #670 (2020-07-22): Upgrading.md - Use file explorer, not alternatives when upgrading on Windows
Documentation update: Add mention of not using alternative file managers / copy handlers, similar to wiki documentation.

Resolves #544
commit 237a1b2d (2020-07-22): Use native file explorer, not alternatives

Add mention of not using alternative file managers / copy handlers, similar to wiki mention.

Resolves #544
closed 2020-07-22 by Ralim