Ralim/IronOS #545: Auto Start feature: True/S?/False, how to set to S?

bl4k1st, 2020-01-01:
The TS100 could autostart the soldering.
The Help text for Auto Start reads T=Soldering, S= Sleep Mode, F=Off

That would mean, True, S= ?, False.

I only have a True/False Checkbox, how can I set S=Sleep?

Ralim, 2020-01-08:
Hmm, 
didn't realise this was missing from the menu, fairly sure the code paths are still there otherwise.
I'll have a dig into this.

ricktendo, 2020-01-18:
Help text is inaccurate, it says the options are: T=Soldering, S= Sleep mode,F=Off

But I only see options "M", "A" and "S"

Also please correct the spacing because this triggers my OCD!

Ralim, 2020-01-18:
@ricktendo Which language did you use? This text is in the translation files, and its also up to the translation file to pick the letter. :) (Your welcome to use the github editor to create a PR if you have time)

ricktendo, 2020-01-18:
@Ralim I am referring to v2.08 English version.

closed 2020-01-18 by Ralim

Nu11u5, 2020-01-22:
Now the behavior seems swapped (based on previous comment).

The help text says "A=Soldering, S=Sleep, M=Off", but the menu options are "S", "F", and "T".

Ralim, 2020-01-22:
@Nu11u5 Please feel free to open a PR to fix the menu text if you get time :)
Its fairly easy as you can edit it on github.com

Firebie, 2020-01-22:
This is already fixed in post-commits after 2.08.1, just not yet released.

colesnicov, 2020-12-21:
https://github.com/Ralim/ts100/issues/635#issuecomment-636218495

PR #558 (2020-01-18): Fix Acting as red light
Please try and fill out this template where possible, not all fields are required and can be removed.

* **Please check if the PR fulfills these requirements**
- [X] The commit message makes sense
- [X] The changes have been tested locally
- [ ] Are there any breaking changes

* **What kind of change does this PR introduce?**
(Bug fix, feature, docs update, ...)
Fix ability to overdrive output and cook the tip into a red light source

Fixes #556 
Fixes #545