Ralim/IronOS #547: Can not find tip model selection setting

linc1990, 2020-01-02:
Hi 
New ts100 owner from Australia here. Just got this nice iron and put your latest firmware on it (2.07) then tried 2.06 28.12.19. I love this firmware but I am having a minor issue. For some reason I can't find the tip selection setting in the menu. I would love to set the tip model and it should be right there in the menu somewhere right so what am I missing! I  have gone through the menu back the front looking for it.  Maybe I just need a nug in the right direction? 
Regards Lincoln 

I'm submitting a bug report

  
  

* **Do you want to request a *feature* or report a *bug*?**

* **What is the current behavior?**

* **What is the expected behavior?**

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1.
2.
3.

***Video of problem if hard to reproduce***

* **What is the motivation / use case for changing the behavior?**

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.06 28-12-19
  - PCB Version: 2
  - Power Supply 19.2 v  3.5 A

* **Other information**

If submitting graphics to go on the iron, please use BMP or PNG files over JPG.

linc1990, 2020-01-05:
(Tip Model Selection

The menu now lets you select which tip you are using with your iron. This lets the unit know which gain value to use for your tip, this varies slightly between models and can help with temperature accuracy. There is also an option for a custom tip. This basically means that you will tune the tip parameters yourself.)

So am I going crazy? Because I can't find this tip model selection anywhere in the menu in the latest 2 versions.

whitehoose, 2020-01-05:
Me too TS80

Firebie, 2020-01-05:
It was removed from firmware.

whitehoose, 2020-01-05:
that would explain it then

linc1990, 2020-01-06:
Why? (Just wondering)

Ralim, 2020-01-06:
Hi,
Indeed as @Firebie has stated it was removed as part of the thermal remodel.

Originally that menu option was there to select different temperature curves for the tips.
During the re-write of all of the temperature handling a few of the errors in the old model were removed which also in my testing removed the requirement for the different curves, ergo the menu only had one option so it was dropped. It will come back if there is a need for it naturally :)

linc1990, 2020-01-06:
Ok nice. I did try to glean that info from all the other threads but couldn't quite put it together. :)

linc1990, 2020-01-06:
Awesome firmware BTW!!! Top stuff this is.

closed 2020-01-06 by linc1990

hwti, 2020-01-06:
> (Tip Model Selection
>
> The menu now lets you select which tip you are using with your iron. This lets the unit know which gain value to use for your tip, this varies slightly between models and can [...]
This comes from https://github.com/Ralim/ts100/blob/master/menu.md#tip-model-selection, this should probably be removed.

Ralim, 2020-01-06:
@hwti Yes it should be, If you have time, can you open a pull request :D