Ralim/IronOS #548: Flash of firmware don't work with the usb device method

RomanPisarew, 2020-01-04:
Hello,

I have issues to flash the firmware with the usb device method. I tried it with the original firmware and this. On Windows I get no storage device in DFU mode, just only in normal mode, where you can change the config. On my Mac I see the storage device and I can copy the firmware, but it just renames the file with a .ERR.

My DFU version is 3.45 and my current firmware is the the stock 2.18.

Are there any other ways to get this firmware in this device?
I got a STM32 Discovery and AFAIK it has a STM Link, can i flash it directly with this? If yes how?

whitehoose, 2020-01-04:
First - splitting hairs I know usually pedantry is of little value, technically - but it might be relevant here.
Stock is what Minidso produce for "their" iron.  Ralim's isn't official so is a "custom" version.
I can only see
Mini's stock v 1.7 http://www.minidso.com/forum.php?mod=viewthread&tid=3208
&
Ralim's custom v 2.07 https://github.com/Ralim/ts100/releases
as being the latest versions 

Is this a slip of the finger - or ????

 I mainly use linux which I imagine is almost Mac and the double dip is normal for me, always has been.
Its an unfortunate process label but like old DOS and early windows - error messages are confusing at best and misleading as usual.

Windows seems to "stick" and often needs manual intervention - but the double dip holds true
For me at least (having done 3rd line support, I'm always cautious and aware that because it works for me means very little globally)

Ralim, 2020-01-04:
@RomanPisarew 
Furthermore to @whitehoose's comments, If you are getting a usb device to let you edit the config, you are running Miniware's firmware (the official one).

As your DFU is 3.45 you are running the slightly newer one that has "bugs fixed" which often makes it less stable on Mac, but more stable on windows. Off memory "stock" 2.18 is moderately old, but i haven't run the stock firmware in quite a while now.

If you are having issues with flashing, searching the GitHub issues here will bring up lots of testing and knowledge, but generally the double-copy method tends to be the best in the _non-windows_ world. (i.e. copy it once, wait for it to show up as .ERR, then copy the same file again, wait for it to show up as both .ERR and .RDY, then power cycle iron).

Alternatively, if you have an st-link, on the TS100 its possible to remove the small carrier board for the stm and flash it directly (careful not to over-write the first 16k which is the USB bootloader). - This should _just work_ if you use the .hex file

RomanPisarew, 2020-01-05:
@whitehoose I forgot to say that is a TS100, I found a stock firmware on this site [http://www.minidso.com/forum.php?mod=viewthread&tid=892&extra=page%3D1](url) and I tried it to flash on linux too.

@Ralim 
I already searched all issues and tried the double copy method, but they don't work on my device.

E.g. with the double copy method I managed it to get a *.rdy file but it was zero bytes so I guess it just an error, because the same name with .err is used so it creates *.rdy file. Anyways after a reboot I had the same version as before.

If I'm flash it with the st-link and I break the bootloader e.g. with overwriting, so I can fix that with a st-link too, because it is not needing the bootloader, right?

whitehoose, 2020-01-05:
I don't ever remember seeing a .rdy and it not completing the update - also can't remember registering a file size - I'll have a look tonight.
I swapped  english for russian last week and it stuck - took about 15 tries before the damn thing changed.  That was windows usually it maybe takes a couple of tries but no more.  I've been playing with some new cheapo esp8266 boards - and they are super fussy about quality of cables - the £1 fleabay specials don't cut it.  I ended up building my own with 0.7mm wire - total overkill but they make good charging leads which shift pixies at a heck of a rate measurably more mA compared to my cheapies, even the ones supplied with "good quality" kit.  I've never had any problems with USB 'till very recently and only with the 8266s but just a thought if yours has somehow developed a taste for fine copper.

whitehoose, 2020-01-05:
ok - Using Linux Mint just shifted to 1.7 stock (17stock.png) then to Ralim's latest.
both versions. Updated as expected ... but on a TS80 - More a case of I started so I'll finish it might help a TS80 owner - and if it is different - I'm interested to know how. 

I've done it twice just to fix the sequence in my mind and check the process.  I'm still using the sexy silicone lead that comes with the TS80
It would be good now if someone with a 100 would check and see if theirs' is the same.

Plug in pressing Button A DFU window opens drop update file into window.
Short pause - beep, window closes, short pause window opens showing *.err file
drop TS80.V1.07.hex update into window (17stock.png) - window shows both files - window freezes.
after a long pause I unplugged ... re-connected DFU - window opened showing 1 file config.txt closed and reopened - .... nothing happened so unplugged ... reconnected normally - iron started - using mini''s ver 1.7 firmware.

Start DFU - window opens, drop Ralim's v2.07 (TS80_en.hex) into window.  Short pause Window opens showing *.err, drop TS80_en.hex into window again (Ralim1.png) - window closes then opens showing . TS80_en.err and TS80_en.rdy - (Ralim2.png)
Window freezes - manual restart Iron starts in Ralim's firmware.

[image]

[image]

[image]

RomanPisarew, 2020-01-05:
@

> I don't ever remember seeing a .rdy and it not completing the update - also can't remember registering a file size - I'll have a look tonight.
> I swapped english for russian last week and it stuck - [...]

It happens when the file was created but has no content. Go to DFU-Mode and touch a file on this device with a hex as suffix, then you will get a *.rdy with zero bytes and nothing happens.

And I flash es8266 board like the wemos d2 mini without problems...
This bootloader is just crap...

RomanPisarew, 2020-01-05:
> ok - Using Linux Mint just shifted to 1.7 stock (17stock.png) then to Ralim's latest.
> both versions. Updated as expected ... but on a TS80 - More a case of I started so I'll finish it might help a [...]

I tried it like this same issue.

whitehoose, 2020-01-05:
I don't know enough about any of this to be smart - I'm just a user   That's the extent of what I've grasped.  With the 100 - I'm not even a user so I'm well out of my depth now.
Hope you crack it

Ralim, 2020-01-08:
@RomanPisarew 
In the last picture it looks like its showing the .RDY file which i would expect it to then load the correct firmware after that ?

In the first image it looks like you have renamed the file to a longer name, this sometimes causes _issues_, but not always.

Also, you could try another USB controller if your using a non intel usb controller, intel (sadly) typically has better compatibility with the out-of-spec timings of the irons.

whitehoose, 2020-01-08:
Both attempts were successful updates - both displayed slight glitchyness but the deed was done
if you see .err - throw another copy in the pot and (for me) that second operation of adding .hex to err is the clincher - even if it seems to halt and you dont see the .rdy a forced restart always gets the update.

The only other thing(s) I can think of are - it's an older machine I multiboot win 7, 10 and linux - to make that work (at the time) involved using standard bios not uefi - I never bothered to alter that.  I don't use a virus scanner on linux I use the free version of kaspersky in both windows - and no third party file manager/file copy helper.  OSs on a multi partition SSD, data stored on hybrid HDD.  I never copy from/to windows system folders on the system partition. (could try running file manager as windows admin) - I don't use root in linux for any of this.
In windows file sharing is via the full old school network access with permissions - I don't let windows home folders control permissions.  I log on with passwords in 7 and pin in 10.

In linux I share using a manually configured fstab

RomanPisarew, 2020-01-08:
@Ralim I tried every thing, can't still update. But can you tell how I connect a st link to the 4 pads?

I found that [image](https://user-images.githubusercontent.com/17366381/29243280-5d4a8ed4-7f9b-11e7-8547-0dfb2c55ac3c.png), but what is SWDO / SWDCLK ... ?

RomanPisarew, 2020-01-12:
@Ralim I found out the mapping of the four pads. I tried it to flash with a st link and a black magic probe. No one works. I can flash other stm32 but not this one... so you can close this issue, I give up...

Ralim, 2020-01-12:
@RomanPisarew Glad you found it, can you please document which pads you used?

If you cant connect, that is quite possible if you are running the stock firmware, as it will disable swd when it starts.

What software are you using for the flashing? Does it support connect SWD based reset triggering ?

RomanPisarew, 2020-01-12:
Hello Ralim,

I connected it like this:

[image]

The first attempt was with a [STM32F4 Discovery](https://www.st.com/en/evaluation-tools/stm32f4discovery.html) and OpenOCD. This board has a st link. I was able to connect but not read or write the flash.

My second attempt was with a [Black Magic Probe](https://github.com/blacksphere/blackmagic/wiki). I flashed the Blackmagic firmware on one of this [STM32 F103 C8T6](https://www.amazon.de/NOSAKA%C2%AE-STM32-F103-C8T6-Arm-STM32-Minimum-Development-Arduino/dp/B01ISYF9U0). I was able to connect and find the uC and attach to it, but again not able to read or write the flash. I tried it with other STM32's and it worked, so the programmer ist not the problem.

> What software are you using for the flashing? Does it support connect SWD based reset triggering ?

I guess yes, but I have not so much experience with the STM32's.

ricktendo, 2020-01-18:
I had the same issue on the latest Windows 10, the solution for me was to use the Linux script.

RomanPisarew, 2020-01-18:
@ricktendo My DFU version is 3.45 and I tried it already with every thing (Linux / Mac / Win7 / Win10).

ricktendo, 2020-01-18:
Bellow you will see the results I obtained using the script, note I used 'sudo su' command to run as root

```
root@nuc:/home/redacted# ./flash_ts100_linux.sh TS100_EN.hex

(process:12203): dconf-WARNING **: 10:32:58.402: failed to commit changes to dconf: Cannot autolaunch D-Bus without X11 $DISPLAY
Found TS100 config disk device on /dev/sdb
Mounted config disk drive, flashing...
'TS100_EN.hex' -> '/tmp/ts100/ts100.hex'
Waiting for TS100 to flash
Remounting config disk drive
Flash result:
/tmp/ts100/ts100.err

(process:12428): dconf-WARNING **: 10:33:04.700: failed to commit changes to dconf: Cannot autolaunch D-Bus without X11 $DISPLAY
root@nuc:/home/redacted# ./flash_ts100_linux.sh TS100_EN.hex

(process:12450): dconf-WARNING **: 10:33:07.202: failed to commit changes to dconf: Cannot autolaunch D-Bus without X11 $DISPLAY
Found TS100 config disk device on /dev/sdc
Mounted config disk drive, flashing...
'TS100_EN.hex' -> '/tmp/ts100/ts100.hex'
Waiting for TS100 to flash
Remounting config disk drive
Flash result:
/tmp/ts100/ts100.err  /tmp/ts100/ts100.rdy

(process:12591): dconf-WARNING **: 10:33:14.950: failed to commit changes to dconf: Cannot autolaunch D-Bus without X11 $DISPLAY
```

grimpod1, 2020-10-12:
@Ralim I have just purchased a TS100 and found operational errors with the stock firmware (version 2.18) and wished to upgrade to your firmware. Unfortunately Windows 10 or Windows 7 will not recognize the TS100 as a virtual disc. The iron correctly displays DF 3.45 on the OLED but that's it. Miniware generously send me another TS100 but it functions exactly the same. I have done the obvious by using a number different micro usb cables but nothing happens. Am I missing something or is there a real problem with the DFU at version 3.45?

Ralim, 2020-10-12:
@grimpod1 
This is most likely your USB controller or software you have installed.
Have you tried multiple computers and multiple USB ports?

grimpod1, 2020-10-12:
@Ralim I have not installed any software on to the controllers. I suspected there may be a hardware problem with my first TS100 controller but the one supplied directly from Miniware made no difference (same firmware version 2.18 -  DFU 3.45) .  I have tried both controllers to be recognized on my Window 10 desk top and Windows 10 Laptop. I even borrowed a friends Laptop with Windows 7 and it is still not recognized. I have ordered another micro usb data sync cable but do not believe that is the problem.

Ralim, 2020-10-12:
@grimpod1 
Can you open device manager when you connect the device?
You should either see it show up as a storage device _or_ show up as an unknown device.
If you see neither its most likely your USB cable. As even broken firmware on the iron will still cause it to show up as an unknown device.

grimpod1, 2020-10-12:
@Ralim I had opened device manager when I connected the device but nothing happens. No icon is displayed as a virtual disc or a message to state an unknown device has been connected. If it was broken firmware I would not expect it to be on both TS100 controllers. I assumed micro USB cables were not just charging cables, but I have sent off for one which is described as data sync as well as charging. It is annoying as the stock firmware (version 2.18) if you change the operation from 'RT' to 'LT' and go back to Standby I can only operate the iron by pressing button B instead of A so I suspect the firmware nesting has gone down the wrong path!

Ralim, 2020-10-12:
>@Ralim I had opened device manager when I connected the device but nothing happens. 

I feel that this is either your pc or the cables you are using I'm afraid

grimpod1, 2020-10-13:
@Ralim I have just purchased a new micro USB cable with data sync and the TS100 controller successfully connected and loaded your firmware into the controller! All my other micro USB cables (5 in all) must be purely be USB charging cables! My mistake I should of thought of the obvious. Thank you for your prompt responses! Have you a user manual for your firmware or are the operational settings sufficiently intuitive not to need one?

Ralim, 2020-10-13:
@grimpod1 

They should be fairly easy.
If you pause on one they have have help text as well.

I encourage you to just use it and experience how it works :)

grimpod1, 2020-10-13:
@Ralim, I had asked the question before I had run through the firmware menu. Is there a way to go back to Standby mode similar to the stock firmware e.g. pressing both A & B buttons?

Ralim, 2020-10-13:
@grimpod1 
Press both buttons or press and hold the rear button (near dc input)

greverfbrayer, 2020-11-13:
> @ grimpod1
> Es muy probable que este sea su controlador USB o software que haya instalado.
> ¿Ha probado varias computadoras y varios puertos USB?

Yes. apparently the bug is with Windows 10, since I tried with Windows 7 and had no problem updating. (Hope to help)

lxofwgktaxl, 2020-11-14:
Just a few words of advice from a very frustrated guy who kept getting .err! Make sure you have a good quality micro usb wire, NO other usb devices plugged in (I was using a laptop) and that the .bin file you are flashing is extracted from the .zip file!

grimpod1, 2020-11-14:
@greverfbrayer If you read my comment to @Ralim above I managed to successfully load Ralim's firmware into my TS100 soldering iron. when I had used a micro USB cable that supported data transfers. The Windows 10 OS recognized the TS100 via the TS100 USB port straight away! My previous Micro USB cables were supplied with USB devices that required charging via USB but none of them, it seems, supported data transfers. The Micro USB cable may good quality but unless all the USB connections are complete (supports data transfers as well as 5V charging) it will not be able to program the firmware into the TS100.

greverfbrayer, 2020-11-15:
@grimpod1 
I need to test the USB cable.
I recently bought one and am waiting for it to arrive.

discip, 2022-04-23:
@rpisarew
This issue seems to be *one of the most popular* if not **the most popular** among issues.
And as @Ralim ***points out every single time***: The solution in most cases is to use another computer, or double copy the hex file, or use another OS.

Please close this issue. 😊

thanks

closed 2022-04-23 by Ralim