Ralim/IronOS #549: Fahrenheit Temperature off by +30

limited660, 2020-01-15:
Please edit this template and fill out all the information you can (where relevant). Failure to provide essential information can delay the response you receive.

* **I'm submitting a ...**
  - [x] Bug report
  - [ ] Feature request
  - [ ] Translation

* **Do you want to request a *feature* or report a *bug*?**
Report a bug

* **What is the current behavior?**
Temperature in fahrenheit reading 30 above everything, displays 100F at 70F room temp and if programmed to 630F it displays 660F, temperature appears correctly when display is in Celsius

* **What is the expected behavior?**
Display the correct temperature

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

***Steps to reproduce:***

1. Turn on iron
2. Switch display to F

* **What are you running:**

On the idle screen, you can hold the settings button and it will show you the firmware version.

  - Firmware Version: 2.7
  - PCB Version: (1/2) - 1.7
  - Power Supply (Voltage and Current Rating) : Anker PowerCore Speed 20000 - 8-10v @ 2.4A

soccerrj88, 2020-03-25:
I have the same problem except mine appears to be off by about 39 degrees. Tried calibration but didn't change it. Tried calibrating while tip was cooled below room temp and it seamed to make it worse. Tried calibrating it while tip was warm and that seemed to maybe make a small difference, but at room temp it is reading around 114F. I noticed as calibration completed each time, the screen flashed a number with a check box before continuing back to the menu. The number was in the 400's with tip cold, around 800 at room temp, and around 1700 with tip warm. Not sure what those numbers indicate, but they were reflecting the change in tip temp.

Ralim, 2020-03-25:
You have a different problem most likely.

Please read the notes about calibration before you use it. The tip and handle must be equal temperature when you calibrate so best means is to leave the iron off for a few hours.

soccerrj88, 2020-03-25:
I should have clarified, I noticed the high tip temp reading when I first plugged the iron in. I have it set to keep the iron off until I press the A button, so the tip should have been the same temp as the body. I followed the instructions on here for calibration to see if that fixed it. When that didn't change anything, then I started playing with the tip temp during calibration to see if that made a difference. That didn't make much difference either, so after letting it sit for a few hours, I re-calibrated it at room temp to bring it back to stock settings and came here to see if there were any notes about the high temp reading. Right now at room temp, 78F, it is currently reading 111F. I have the TS-I tip.

Ralim, 2020-12-30:
Apologies on this slipping through the cracks 😢 

I _believe_ this should now be fixed.
If you get some time, it would be fantastic if you check the latest CI build to test this (currently [this one](https://github.com/Ralim/ts100/actions/runs/452337083)).

Just scroll down and click on your model to download a zip of all of the languages.
🙏🏼

discip, 2022-04-23:
@limited660
Have you tried the latest release yet?
If this solves your issue please don't forget to close it. 😊

thanks

soccerrj88, 2022-04-24:
I believe this is fixed now. It is showing the correct room temp when I turn it on. Thank you!

soccerrj88, 2022-04-24:
Ok, I don't see how to close this issue, but it can be closed out.

closed 2022-04-24 by Ralim